llvm/circt #7904: [Nightly] Transforms/switch-to-if.mlir fails valgrind test

teqdruid, 2024-11-26:
```
******************** TEST 'CIRCT :: Transforms/switch-to-if.mlir' FAILED ********************
Exit Code: 123

Command Output (stderr):
--
RUN: at line 1: /__w/circt/circt/build/bin/circt-opt -split-input-file --switch-to-if /__w/circt/circt/test/Transforms/switch-to-if.mlir | /__w/circt/circt/build/bin/FileCheck /__w/circt/circt/test/Transforms/switch-to-if.mlir
+ /__w/circt/circt/build/bin/circt-opt -split-input-file --switch-to-if /__w/circt/circt/test/Transforms/switch-to-if.mlir
+ /__w/circt/circt/build/bin/FileCheck /__w/circt/circt/test/Transforms/switch-to-if.mlir
==31750== Invalid read of size 8
==31750==    at 0x2BF56A0: mlir::Block::getParentOp() (in /__w/circt/circt/build/bin/circt-opt)
==31750==    by 0x36ACFD8: llvm::LogicalResult llvm::function_ref<llvm::LogicalResult (mlir::Pattern const&)>::callback_fn<(anonymous namespace)::OperationLegalizer::legalizeWithPattern(mlir::Operation*, mlir::ConversionPatternRewriter&)::$_2>(long, mlir::Pattern const&) (in /__w/circt/circt/build/bin/circt-opt)
==31750==    by 0x36E77C0: void llvm::function_ref<void ()>::callback_fn<mlir::PatternApplicator::matchAndRewrite(mlir::Operation*, mlir::PatternRewriter&, llvm::function_ref<bool (mlir::Pattern const&)>, llvm::function_ref<void (mlir::Pattern const&)>, llvm::function_ref<llvm::LogicalResult (mlir::Pattern const&)>)::$_0>(long) (in /__w/circt/circt/build/bin/circt-opt)
==31750==    by 0x36E462E: mlir::PatternApplicator::matchAndRewrite(mlir::Operation*, mlir::PatternRewriter&, llvm::function_ref<bool (mlir::Pattern const&)>, llvm::function_ref<void (mlir::Pattern const&)>, llvm::function_ref<llvm::LogicalResult (mlir::Pattern const&)>) (in /__w/circt/circt/build/bin/circt-opt)
==31750==    by 0x36A276C: (anonymous namespace)::OperationLegalizer::legalize(mlir::Operation*, mlir::ConversionPatternRewriter&) (in /__w/circt/circt/build/bin/circt-opt)
==31750==    by 0x36A1806: mlir::OperationConverter::convert(mlir::ConversionPatternRewriter&, mlir::Operation*) (in /__w/circt/circt/build/bin/circt-opt)
==31750==    by 0x36A29AE: mlir::OperationConverter::convertOperations(llvm::ArrayRef<mlir::Operation*>) (in /__w/circt/circt/build/bin/circt-opt)
==31750==    by 0x36AA2BA: mlir::applyPartialConversion(mlir::Operation*, mlir::ConversionTarget const&, mlir::FrozenRewritePatternSet const&, mlir::ConversionConfig) (in /__w/circt/circt/build/bin/circt-opt)
==31750==    by 0x2BBA2C4: (anonymous namespace)::IndexSwitchToIfPass::runOnOperation() (in /__w/circt/circt/build/bin/circt-opt)
==31750==    by 0x3840D05: mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) (in /__w/circt/circt/build/bin/circt-opt)
==31750==    by 0x3841531: mlir::detail::OpToOpPassAdaptor::runPipeline(mlir::OpPassManager&, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) (in /__w/circt/circt/build/bin/circt-opt)
==31750==    by 0x3843D5D: mlir::PassManager::run(mlir::Operation*) (in /__w/circt/circt/build/bin/circt-opt)
==31750==  Address 0x69074a8 is 24 bytes inside a block of size 72 free'd
==31750==    at 0x483CFBF: operator delete(void*) (in /usr/lib/x86_64-linux-gnu/valgrind/vgpreload_memcheck-amd64-linux.so)
==31750==    by 0x2BBAF7E: SwitchToIfConversion::matchAndRewrite(mlir::scf::IndexSwitchOp, mlir::scf::IndexSwitchOpAdaptor, mlir::ConversionPatternRewriter&) const (in /__w/circt/circt/build/bin/circt-opt)
==31750==    by 0x2BBAB5F: mlir::OpConversionPattern<mlir::scf::IndexSwitchOp>::matchAndRewrite(mlir::Operation*, llvm::ArrayRef<mlir::Value>, mlir::ConversionPatternRewriter&) const (in /__w/circt/circt/build/bin/circt-opt)
==31750==    by 0x36A1770: mlir::ConversionPattern::matchAndRewrite(mlir::Operation*, mlir::PatternRewriter&) const (in /__w/circt/circt/build/bin/circt-opt)
==31750==    by 0x36E79E9: void llvm::function_ref<void ()>::callback_fn<mlir::PatternApplicator::matchAndRewrite(mlir::Operation*, mlir::PatternRewriter&, llvm::function_ref<bool (mlir::Pattern const&)>, llvm::function_ref<void (mlir::Pattern const&)>, llvm::function_ref<llvm::LogicalResult (mlir::Pattern const&)>)::$_0>(long) (in /__w/circt/circt/build/bin/circt-opt)
==31750==    by 0x36E462E: mlir::PatternApplicator::matchAndRewrite(mlir::Operation*, mlir::PatternRewriter&, llvm::function_ref<bool (mlir::Pattern const&)>, llvm::function_ref<void (mlir::Pattern const&)>, llvm::function_ref<llvm::LogicalResult (mlir::Pattern const&)>) (in /__w/circt/circt/build/bin/circt-opt)
==31750==    by 0x36A276C: (anonymous namespace)::OperationLegalizer::legalize(mlir::Operation*, mlir::ConversionPatternRewriter&) (in /__w/circt/circt/build/bin/circt-opt)
==31750==    by 0x36A1806: mlir::OperationConverter::convert(mlir::ConversionPatternRewriter&, mlir::Operation*) (in /__w/circt/circt/build/bin/circt-opt)
==31750==    by 0x36A29AE: mlir::OperationConverter::convertOperations(llvm::ArrayRef<mlir::Operation*>) (in /__w/circt/circt/build/bin/circt-opt)
==31750==    by 0x36AA2BA: mlir::applyPartialConversion(mlir::Operation*, mlir::ConversionTarget const&, mlir::FrozenRewritePatternSet const&, mlir::ConversionConfig) (in /__w/circt/circt/build/bin/circt-opt)
==31750==    by 0x2BBA2C4: (anonymous namespace)::IndexSwitchToIfPass::runOnOperation() (in /__w/circt/circt/build/bin/circt-opt)
==31750==    by 0x3840D05: mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) (in /__w/circt/circt/build/bin/circt-opt)
==31750==  Block was alloc'd at
==31750==    at 0x483BE63: operator new(unsigned long) (in /usr/lib/x86_64-linux-gnu/valgrind/vgpreload_memcheck-amd64-linux.so)
==31750==    by 0x2BF96E3: mlir::OpBuilder::createBlock(mlir::Region*, llvm::ilist_iterator<llvm::ilist_detail::node_options<mlir::Block, true, false, void, false, void>, false, false>, mlir::TypeRange, llvm::ArrayRef<mlir::Location>) (in /__w/circt/circt/build/bin/circt-opt)
==31750==    by 0x310E77A: mlir::scf::IfOp::build(mlir::OpBuilder&, mlir::OperationState&, mlir::TypeRange, mlir::Value, bool) (in /__w/circt/circt/build/bin/circt-opt)
==31750==    by 0x2A6D60A: mlir::scf::IfOp mlir::OpBuilder::create<mlir::scf::IfOp, mlir::ValueTypeRange<mlir::ResultRange>, mlir::Value&, bool>(mlir::Location, mlir::ValueTypeRange<mlir::ResultRange>&&, mlir::Value&, bool&&) (in /__w/circt/circt/build/bin/circt-opt)
==31750==    by 0x2BBAEE1: SwitchToIfConversion::matchAndRewrite(mlir::scf::IndexSwitchOp, mlir::scf::IndexSwitchOpAdaptor, mlir::ConversionPatternRewriter&) const (in /__w/circt/circt/build/bin/circt-opt)
==31750==    by 0x2BBAB5F: mlir::OpConversionPattern<mlir::scf::IndexSwitchOp>::matchAndRewrite(mlir::Operation*, llvm::ArrayRef<mlir::Value>, mlir::ConversionPatternRewriter&) const (in /__w/circt/circt/build/bin/circt-opt)
==31750==    by 0x36A1770: mlir::ConversionPattern::matchAndRewrite(mlir::Operation*, mlir::PatternRewriter&) const (in /__w/circt/circt/build/bin/circt-opt)
==31750==    by 0x36E79E9: void llvm::function_ref<void ()>::callback_fn<mlir::PatternApplicator::matchAndRewrite(mlir::Operation*, mlir::PatternRewriter&, llvm::function_ref<bool (mlir::Pattern const&)>, llvm::function_ref<void (mlir::Pattern const&)>, llvm::function_ref<llvm::LogicalResult (mlir::Pattern const&)>)::$_0>(long) (in /__w/circt/circt/build/bin/circt-opt)
==31750==    by 0x36E462E: mlir::PatternApplicator::matchAndRewrite(mlir::Operation*, mlir::PatternRewriter&, llvm::function_ref<bool (mlir::Pattern const&)>, llvm::function_ref<void (mlir::Pattern const&)>, llvm::function_ref<llvm::LogicalResult (mlir::Pattern const&)>) (in /__w/circt/circt/build/bin/circt-opt)
==31750==    by 0x36A276C: (anonymous namespace)::OperationLegalizer::legalize(mlir::Operation*, mlir::ConversionPatternRewriter&) (in /__w/circt/circt/build/bin/circt-opt)
==31750==    by 0x36A1806: mlir::OperationConverter::convert(mlir::ConversionPatternRewriter&, mlir::Operation*) (in /__w/circt/circt/build/bin/circt-opt)
==31750==    by 0x36A29AE: mlir::OperationConverter::convertOperations(llvm::ArrayRef<mlir::Operation*>) (in /__w/circt/circt/build/bin/circt-opt)
==31750== 

--
```

teqdruid, 2024-11-26:
Oops. @seldridge already reverted the offending commit https://github.com/llvm/circt/commit/b987492739855b3d508922bbebb84e12f2649e8f

closed 2024-11-26 by teqdruid

seldridge, 2024-11-26:
Thanks for the report, @teqdruid.

I have an internal team devops channel setup to monitor the nightlies, so I will usually know when there is a failure. With the revert cherry-picked, nightly passed here (https://github.com/llvm/circt/actions/runs/12035244286). I also verified this on a local machine.

teqdruid, 2024-11-26:
I was wondering how you kept on top of the nightly runs so quickly! I think I'm still the only one getting the notification emails on failures directly from GH.