llvm/circt #7957: FIRRTL memory read-under-write is not respected

programmerjake, 2024-12-06:
I tried generating verilog for both `old` and `new` and the generated verilog is identical, which is incorrect.
```
firtool --version
LLVM (http://llvm.org/):
  LLVM version 20.0.0git
  Optimized build.
CIRCT firtool-1.96.0
```
I used `firtool --preserve-values=all --mlir-print-ir-after-all --mlir-print-ir-after-change memory.fir` and the difference disappeared after the `HWMemSimImpl` (`hw-memory-sim`) pass.

`old` FIRRTL:
```firrtl
FIRRTL version 3.2.0
circuit memory:
    module memory:
        input r0: {addr: UInt<8>, en: UInt<1>, clk: Clock, flip data: UInt<8>[2]}
        input w1: {addr: UInt<8>, en: UInt<1>, clk: Clock, data: UInt<8>[2], mask: UInt<1>[2]}
        mem memory:
            data-type => UInt<8>[2]
            depth => 256
            read-latency => 1
            write-latency => 1
            read-under-write => old
            reader => r0
            writer => w1
        connect memory.r0, r0
        connect memory.w1, w1
```

`new` FIRRTL:
```firrtl
FIRRTL version 3.2.0
circuit memory:
    module memory:
        input r0: {addr: UInt<8>, en: UInt<1>, clk: Clock, flip data: UInt<8>[2]}
        input w1: {addr: UInt<8>, en: UInt<1>, clk: Clock, data: UInt<8>[2], mask: UInt<1>[2]}
        mem memory:
            data-type => UInt<8>[2]
            depth => 256
            read-latency => 1
            write-latency => 1
            read-under-write => new
            reader => r0
            writer => w1
        connect memory.r0, r0
        connect memory.w1, w1
```

seldridge, 2024-12-06:
Sorry about this. This is an old duplicate of: https://github.com/llvm/circt/issues/4588 We just never implemented this.

I don't think this feature is that difficult to implement if you'd like to take a look at it, it just hasn't been a priority. (This was also unreachable for a time from Chisel, though it is entirely reachable now.)

programmerjake, 2024-12-06:
ok, closing in favor of #4588

closed 2024-12-06 by programmerjake