llvm/circt #7970: [CAPI] Windows and Nightly Failure after #7960

seldridge, 2024-12-11:
I'm going to blamelessly revert #7960 (https://github.com/llvm/circt/commit/14622691f3c63793411a38b3fa874cb6866c9f43). Please re-land this @SpriteOvO, once you've fixed the issues with it.

Here's what I'm seeing on my end from `valgrind`:

```
# valgrind --track-origins=yes /home/schuylere/scratch/repos/github.com/llvm/circt/build-clang-clang++-RelWithDebInfo-on-vg/bin/circt-capi-firrtl-test
==3842745== Memcheck, a memory error detector
==3842745== Copyright (C) 2002-2022, and GNU GPL'd, by Julian Seward et al.
==3842745== Using Valgrind-3.21.0 and LibVEX; rerun with -h for copyright info
==3842745== Command: /home/schuylere/scratch/repos/github.com/llvm/circt/build-clang-clang++-RelWithDebInfo-on-vg/bin/circt-capi-firrtl-test
==3842745==
FIRRTL version 4.1.0
circuit ExportTestSimpleModule :
  public module ExportTestSimpleModule : @[- 2:3]
    input in_1 : UInt<32> @[- 2:44]
    input in_2 : UInt<32> @[- 3:44]
    output out : UInt<32> @[- 4:45]

    connect out, and(in_1, in_2) @[- 6:5]
module {
  firrtl.circuit "AnnoTest" attributes {rawAnnotations = [{class = "firrtl.transforms.DontTouchAnnotation", target = "~AnnoTest|AnnoTest>in"}]} {
    firrtl.module @AnnoTest(in %in: !firrtl.uint<32>) {
    }
  }
}
==3842745== Conditional jump or move depends on uninitialised value(s)
==3842745==    at 0x405B30: testTypeDiscriminantsAndQueries (test/CAPI/firrtl.c:300)
==3842745==    by 0x4065C4: main (test/CAPI/firrtl.c:368)
==3842745==  Uninitialised value was created by a stack allocation
==3842745==    at 0x4053BE: testTypeDiscriminantsAndQueries (test/CAPI/firrtl.c:192)
==3842745==
==3842745== Conditional jump or move depends on uninitialised value(s)
==3842745==    at 0x405B7D: testTypeDiscriminantsAndQueries (test/CAPI/firrtl.c:302)
==3842745==    by 0x4065C4: main (test/CAPI/firrtl.c:368)
==3842745==  Uninitialised value was created by a stack allocation
==3842745==    at 0x409A60: firrtlTypeGetBundle (lib/CAPI/Dialect/FIRRTL.cpp:119)
==3842745==
==3842745==
==3842745== HEAP SUMMARY:
==3842745==     in use at exit: 388,477 bytes in 2,575 blocks
==3842745==   total heap usage: 3,135 allocs, 560 frees, 616,958 bytes allocated
==3842745==
==3842745== LEAK SUMMARY:
==3842745==    definitely lost: 224 bytes in 4 blocks
==3842745==    indirectly lost: 315,133 bytes in 2,566 blocks
==3842745==      possibly lost: 208 bytes in 2 blocks
==3842745==    still reachable: 72,912 bytes in 3 blocks
==3842745==         suppressed: 0 bytes in 0 blocks
==3842745== Rerun with --leak-check=full to see details of leaked memory
==3842745==
==3842745== For lists of detected and suppressed errors, rerun with: -s
==3842745== ERROR SUMMARY: 2 errors from 2 contexts (suppressed: 0 from 0)
```

closed 2024-12-13 by SpriteOvO

seldridge, 2024-12-13:
Thanks for taking care of this @SpriteOvO!

SpriteOvO, 2024-12-13:
@seldridge I remember I have seen several times this kind of Windows only error after PRs merged, what do you think if we enable Windows builds CI for each commits for all branches rather than only for main branch?

SpriteOvO, 2024-12-13:
CC @teqdruid for changing it to only main branch before, what do you think that?

seldridge, 2024-12-13:
SGTM. The Windows builds are approximately the time or slightly faster than the normal builds. This would catch a number of problems that have been coming up recently.

I can't recall if we used to have these on PRs and if we turned them off for some reason.

teqdruid, 2024-12-13:
Yeah, it used to be that the Windows builds took significantly longer than the Linux builds. But I've been thinking about this myself recently so I've no objection to enabling Windows builds on PRs.

Now if we could just get the integration tests working on Windows...