llvm/circt #7984: Build and test LLVM/MLIR Failed

quanchenliu, 2024-12-13:
**What I did before:**
```
$ git clone git@github.com:llvm/circt.git
$ cd circt
$ git submodule init
$ git submodule update
$ mkdir llvm/build
$ cd llvm/build
$ cmake -G Ninja ../llvm \
    -DLLVM_ENABLE_PROJECTS="mlir" \
    -DLLVM_TARGETS_TO_BUILD="host" \
    -DLLVM_ENABLE_ASSERTIONS=ON \
    -DCMAKE_BUILD_TYPE=DEBUG \
    -DCMAKE_EXPORT_COMPILE_COMMANDS=ON
$ ninja
[1415/4773] Building CXX object lib/Transforms/Vectoriz...iles/LLVMVectorize.dir/SandboxVectorizer/Legality.cpp.o
In file included from /home/lqc/circt/llvm/llvm/lib/Transforms/Vectorize/SandboxVectorizer/Legality.cpp:9:
/home/lqc/circt/llvm/llvm/include/llvm/Transforms/Vectorize/SandboxVectorizer/Legality.h:115:33: warning: ‘maybe_unused’ attribute ignored [-Wattributes]
  115 |   [[maybe_unused]] ResultReason Reason;
      |                                 ^~~~~~
[1417/4773] Building CXX object lib/Transforms/Vectoriz...ectorize.dir/SandboxVectorizer/Passes/BottomUpVec.cpp.o
In file included from /home/lqc/circt/llvm/llvm/include/llvm/Transforms/Vectorize/SandboxVectorizer/Passes/BottomUpVec.h:21,
                 from /home/lqc/circt/llvm/llvm/lib/Transforms/Vectorize/SandboxVectorizer/Passes/BottomUpVec.cpp:9:
/home/lqc/circt/llvm/llvm/include/llvm/Transforms/Vectorize/SandboxVectorizer/Legality.h:115:33: warning: ‘maybe_unused’ attribute ignored [-Wattributes]
  115 |   [[maybe_unused]] ResultReason Reason;
      |                                 ^~~~~~
[1418/4773] Building CXX object lib/Transforms/Vectoriz...ir/SandboxVectorizer/SandboxVectorizerPassBuilder.cpp.o
In file included from /home/lqc/circt/llvm/llvm/include/llvm/Transforms/Vectorize/SandboxVectorizer/Passes/BottomUpVec.h:21,
                 from /home/lqc/circt/llvm/llvm/lib/Transforms/Vectorize/SandboxVectorizer/SandboxVectorizerPassBuilder.cpp:3:
/home/lqc/circt/llvm/llvm/include/llvm/Transforms/Vectorize/SandboxVectorizer/Legality.h:115:33: warning: ‘maybe_unused’ attribute ignored [-Wattributes]
  115 |   [[maybe_unused]] ResultReason Reason;
      |                                 ^~~~~~
[1433/4773] Building CXX object lib/Transforms/Vectorize/CMakeFiles/LLVMVectorize.dir/VPlanUtils.cpp.o
In file included from /home/lqc/circt/llvm/llvm/lib/Transforms/Vectorize/VPlanUtils.cpp:10:
/home/lqc/circt/llvm/llvm/lib/Transforms/Vectorize/VPlanPatternMatch.h: In instantiation of ‘bool llvm::VPlanPatternMatch::detail::CheckTupleElements(const TupleTy&, Fn, std::index_sequence<_Idx ...>) [with TupleTy = std::tuple<>; Fn = llvm::VPlanPatternMatch::Recipe_match<std::tuple<>, 0, false, llvm::VPCanonicalIVPHIRecipe>::match(const llvm::VPRecipeBase*) const::<lambda(auto:28, unsigned int)>; long unsigned int ...Is = {}; std::index_sequence<_Idx ...> = std::integer_sequence<long unsigned int>]’:
/home/lqc/circt/llvm/llvm/lib/Transforms/Vectorize/VPlanPatternMatch.h:161:28:   required from ‘bool llvm::VPlanPatternMatch::detail::all_of_tuple_elements(const TupleTy&, Fn) [with TupleTy = std::tuple<>; Fn = llvm::VPlanPatternMatch::Recipe_match<std::tuple<>, 0, false, llvm::VPCanonicalIVPHIRecipe>::match(const llvm::VPRecipeBase*) const::<lambda(auto:28, unsigned int)>]’
/home/lqc/circt/llvm/llvm/lib/Transforms/Vectorize/VPlanPatternMatch.h:197:38:   required from ‘bool llvm::VPlanPatternMatch::Recipe_match<Ops_t, Opcode, Commutative, RecipeTys>::match(const llvm::VPRecipeBase*) const [with Ops_t = std::tuple<>; unsigned int Opcode = 0; bool Commutative = false; RecipeTys = {llvm::VPCanonicalIVPHIRecipe}]’
/home/lqc/circt/llvm/llvm/lib/Transforms/Vectorize/VPlanPatternMatch.h:184:25:   required from ‘bool llvm::VPlanPatternMatch::Recipe_match<Ops_t, Opcode, Commutative, RecipeTys>::match(const llvm::VPValue*) const [with Ops_t = std::tuple<>; unsigned int Opcode = 0; bool Commutative = false; RecipeTys = {llvm::VPCanonicalIVPHIRecipe}]’
/home/lqc/circt/llvm/llvm/lib/Transforms/Vectorize/VPlanPatternMatch.h:198:45:   required from ‘llvm::VPlanPatternMatch::Recipe_match<std::tuple<llvm::VPlanPatternMatch::Recipe_match<std::tuple<>, 0, false, llvm::VPCanonicalIVPHIRecipe>, llvm::VPlanPatternMatch::specific_intval<0> >, 0, false, llvm::VPScalarIVStepsRecipe>::match(const llvm::VPRecipeBase*) const::<lambda(auto:28, unsigned int)> [with auto:28 = llvm::VPlanPatternMatch::Recipe_match<std::tuple<>, 0, false, llvm::VPCanonicalIVPHIRecipe>]’
/home/lqc/circt/llvm/llvm/lib/Transforms/Vectorize/VPlanPatternMatch.h:155:12:   required from ‘bool llvm::VPlanPatternMatch::detail::CheckTupleElements(const TupleTy&, Fn, std::index_sequence<_Idx ...>) [with TupleTy = std::tuple<llvm::VPlanPatternMatch::Recipe_match<std::tuple<>, 0, false, llvm::VPCanonicalIVPHIRecipe>, llvm::VPlanPatternMatch::specific_intval<0> >; Fn = llvm::VPlanPatternMatch::Recipe_match<std::tuple<llvm::VPlanPatternMatch::Recipe_match<std::tuple<>, 0, false, llvm::VPCanonicalIVPHIRecipe>, llvm::VPlanPatternMatch::specific_intval<0> >, 0, false, llvm::VPScalarIVStepsRecipe>::match(const llvm::VPRecipeBase*) const::<lambda(auto:28, unsigned int)>; long unsigned int ...Is = {0, 1}; std::index_sequence<_Idx ...> = std::integer_sequence<long unsigned int, 0, 1>]’
/home/lqc/circt/llvm/llvm/lib/Transforms/Vectorize/VPlanPatternMatch.h:161:28:   required from ‘bool llvm::VPlanPatternMatch::detail::all_of_tuple_elements(const TupleTy&, Fn) [with TupleTy = std::tuple<llvm::VPlanPatternMatch::Recipe_match<std::tuple<>, 0, false, llvm::VPCanonicalIVPHIRecipe>, llvm::VPlanPatternMatch::specific_intval<0> >; Fn = llvm::VPlanPatternMatch::Recipe_match<std::tuple<llvm::VPlanPatternMatch::Recipe_match<std::tuple<>, 0, false, llvm::VPCanonicalIVPHIRecipe>, llvm::VPlanPatternMatch::specific_intval<0> >, 0, false, llvm::VPScalarIVStepsRecipe>::match(const llvm::VPRecipeBase*) const::<lambda(auto:28, unsigned int)>]’
/home/lqc/circt/llvm/llvm/lib/Transforms/Vectorize/VPlanPatternMatch.h:197:38:   required from ‘bool llvm::VPlanPatternMatch::Recipe_match<Ops_t, Opcode, Commutative, RecipeTys>::match(const llvm::VPRecipeBase*) const [with Ops_t = std::tuple<llvm::VPlanPatternMatch::Recipe_match<std::tuple<>, 0, false, llvm::VPCanonicalIVPHIRecipe>, llvm::VPlanPatternMatch::specific_intval<0> >; unsigned int Opcode = 0; bool Commutative = false; RecipeTys = {llvm::VPScalarIVStepsRecipe}]’
/home/lqc/circt/llvm/llvm/lib/Transforms/Vectorize/VPlanPatternMatch.h:184:25:   required from ‘bool llvm::VPlanPatternMatch::Recipe_match<Ops_t, Opcode, Commutative, RecipeTys>::match(const llvm::VPValue*) const [with Ops_t = std::tuple<llvm::VPlanPatternMatch::Recipe_match<std::tuple<>, 0, false, llvm::VPCanonicalIVPHIRecipe>, llvm::VPlanPatternMatch::specific_intval<0> >; unsigned int Opcode = 0; bool Commutative = false; RecipeTys = {llvm::VPScalarIVStepsRecipe}]’
/home/lqc/circt/llvm/llvm/lib/Transforms/Vectorize/VPlanPatternMatch.h:29:17:   required from ‘bool llvm::VPlanPatternMatch::match(Val*, const Pattern&) [with Val = llvm::VPValue; Pattern = llvm::VPlanPatternMatch::Recipe_match<std::tuple<llvm::VPlanPatternMatch::Recipe_match<std::tuple<>, 0, false, llvm::VPCanonicalIVPHIRecipe>, llvm::VPlanPatternMatch::specific_intval<0> >, 0, false, llvm::VPScalarIVStepsRecipe>]’
/home/lqc/circt/llvm/llvm/lib/Transforms/Vectorize/VPlanUtils.cpp:58:18:   required from here
/home/lqc/circt/llvm/llvm/lib/Transforms/Vectorize/VPlanPatternMatch.h:154:48: warning: parameter ‘P’ set but not used [-Wunused-but-set-parameter]
  154 | bool CheckTupleElements(const TupleTy &Ops, Fn P, std::index_sequence<Is...>) {
      |                                             ~~~^
/home/lqc/circt/llvm/llvm/lib/Transforms/Vectorize/VPlanPatternMatch.h: In instantiation of ‘bool llvm::VPlanPatternMatch::detail::CheckTupleElements(const TupleTy&, Fn, std::index_sequence<_Idx ...>) [with TupleTy = std::tuple<>; Fn = llvm::VPlanPatternMatch::Recipe_match<std::tuple<>, 0, false, llvm::VPCanonicalIVPHIRecipe>::match(const llvm::VPRecipeBase*) const::<lambda(auto:29, unsigned int)>; long unsigned int ...Is = {}; std::index_sequence<_Idx ...> = std::integer_sequence<long unsigned int>]’:
/home/lqc/circt/llvm/llvm/lib/Transforms/Vectorize/VPlanPatternMatch.h:161:28:   required from ‘bool llvm::VPlanPatternMatch::detail::all_of_tuple_elements(const TupleTy&, Fn) [with TupleTy = std::tuple<>; Fn = llvm::VPlanPatternMatch::Recipe_match<std::tuple<>, 0, false, llvm::VPCanonicalIVPHIRecipe>::match(const llvm::VPRecipeBase*) const::<lambda(auto:29, unsigned int)>]’
[... cut ...]
ninja: build stopped: subcommand failed.
```
**My Configuration**: 
```
Operation System：Ubuntu-22.04（WSL2）
cmake version 3.22.1
Python 3.10.12
zlib 1.3.1
GNU Make 4.3
PyYAML 6.0.2
```
**My Question**：
First, I would like to know if the content in the "Setting this up" section of the README needs to be executed in order. Specifically, does the LLVM/MLIR build and test need to be completed first before successfully building CIRCT?
Additionally, I would like to ask how to resolve issues encountered when executing the ninja command.

maerhart, 2024-12-13:
> First, I would like to know if the content in the "Setting this up" section of the README needs to be executed in order. Specifically, does the LLVM/MLIR build and test need to be completed first before successfully building CIRCT?

Yes, the steps need to be followed in order. CIRCT links against LLVM/MLIR and thus the libraries must be in the llvm build folder at the time you compile CIRCT.

The kind of failure you're getting is often the result of running out of memory (Linux just kills apps if you use too much memory). If you're using `ld` as linker that might be the problem. I'd suggest to use `lld` or any other linker.
You can also try adding the `-j 1` flag to ninja to only compile on one thread, thus reducing peak memory consumption.

AndreCorreaSantos, 2025-01-10:
Hello, following up on this topic, If I have recently built llvm and mlir in some other location, can I use that built version to run CIRCT? Or do I have to build this specific version that is present as a submodule in the repo?

hamidelmaazouz, 2025-02-05:
Same here, I have a similar situation, I have latest LLVM already built, is there an easy way to just link against that instead of as a git submodule ?

hamidelmaazouz, 2025-02-05:
I suppose I could just tinker with CMake variables. Usually it's `LLVM_DIR` but the cmake modules seem to be cohesively tied to the llvm submodule