llvm/circt #8002: [Seq] Counters

teqdruid, 2024-12-17:
I find myself wanting some counter ops. Right now, it'd be _very_ convenient to have a rollover counter op. Is the seq dialect the place to put this? Or is it too high-level? If it's not the correct dialect, which one is?

teqdruid, 2024-12-17:
We've been discussing a mid-level dialect with common IP/ops forever. Does this belong there?

darthscsi, 2025-01-14:
I've always figured there needs to be a "library" dialect for common stuff like this.  It's trivially lowerable to comb+seq, so it probably shouldn't be in either of those.

teqdruid, 2025-01-15:
> I've always figured there needs to be a "library" dialect for common stuff like this. It's trivially lowerable to comb+seq, so it probably shouldn't be in either of those.

I don't necessarily disagree with this, but counterpoint: the truth table op in comb gets lowered into muxes pretty trivially.