llvm/circt #8012: [Moore][Arc][LLHD] Moore to LLVM lowering issues

likeamahoney, 2024-12-20:
Hi all!

I am trying to simulate a SystemVerilog code (listed below) using `arcilator`:

```verilog
module dff(D, clk, Q);
    input D; // Data input 
    input clk; // clock input 
    output reg Q; // output Q

    // Simple DFF
    always @(negedge clk) 
    begin
        Q <= D; 
    end 
endmodule
```

 with such pipeline:

```bash
circt-verilog dff.sv | arcilator
```

`arcilator` fails with  **error: 'llhd.process' op has regions; not supported by ConvertToArcs**

Then I try to use a different pipeline with `circt-opt` :

```bash
circt-verilog dff.sv | circt-opt --llhd-early-code-motion --llhd-temporal-code-motion --llhd-desequentialize --llhd-sig2reg -canonicalize | arcilator
```

`acilator` fails with another **error: failed to legalize operation 'seq.clock_inv'**

What is necessary to be able to run `arcilator` on such simple `DFF` example?

maerhart, 2024-12-21:
Hi @likeamahoney 

Thanks for checking out SystemVerilog simulation with Arcilator!

This lowering path is still in early development and the issue you are encountering is because of using `negedge` while we currently only support `posedge` upstream (because that's what Chisel/FIRRTL uses). The fix is simple: just add a lowering pattern for `seq.clock_inv` to `LowerArcToLLVM.cpp`.

I have [a branch](https://github.com/maerhart/circt/tree/mix-and-match) with some fixes and additional features (including that one) that should be enough to fully lower your example, if you want to give it a try. I hope to upstream that sometime soon. On the branch, you can simply run arcilator directly after `circt-verilog` because the LLHD pipeline was added to `circt-verilog`.

Thanks for opening this issue, it's very useful to see what issues users run into 😄

likeamahoney, 2024-12-21:
> Hi [@likeamahoney](https://github.com/likeamahoney)
>
> Thanks for checking out SystemVerilog simulation with Arcilator!
>
> This lowering path is still in early development and the issue you are [...]

Hi! 

Much thanks! I tested your branch and I managed to translate all my examples related to triggers and sequential logic.

Hope it would be merged in main branch soon! 

But combinational logic does not translate so well.

Combinational examples with continuous assignments  only are lowered correctly:

For example:

```verilog
module combo (  input   a, b, c, d, e, output  z);
        assign z = ((a & b) | (c ^ d) & ~e);
endmodule
```

But `arcilator` emits an error (`error: 'llhd.process' op has regions; not supported by ConvertToArcs`) when combinational logic is inside an `always` block. For example:

```verilog
module combo (  input   a, b, c, d, e,  output  reg z);

        always @ (*) begin
                z = ((a & b) | (c ^ d) & ~e);
        end

endmodule
```

likeamahoney, 2024-12-22:
> Thanks for pointing this out. I created an issue tracking this: [#8013](https://github.com/llvm/circt/issues/8013)
> 
> In the meantime, you can use `always_comb` instead.

Thanks a lot!

Can I leave this issue open for future lowering issues reports or should I to close it?