llvm/circt #8022: [Comb] Infinite loop in OrOp folder

maerhart, 2024-12-31:
The following IR ends up in an infinite loop in the `comb.or` folder. It's a degenerate case, but the compiler should still not end up in an infinite loop. I think we can just leave the IR as is or fold it to some kind of poison value.
```mlir
hw.module @infiniteFolding(out out : i1) {
  %true = hw.constant true
  %0 = comb.mux %0, %true, %0 : i1
  hw.output %0 : i1
}
```

With `--debug-only=greedy-rewriter`:
```
//===-------------------------------------------===//
Processing operation : 'comb.or'(0x561b88d1c530) {
  %1 = "comb.or"(%1, %1) : (i1, i1) -> i1

} -> success : operation was folded
//===-------------------------------------------===//
```

asmit27rai, 2025-01-21:
@maerhart Can you pls assign this issue to me. I want to work on this issue.

maerhart, 2025-01-23:
Thanks for working on it @asmit27rai !