llvm/circt #8030: [Feature Request][FIRRTL] add a pass on firrtl.circuit to create a firrtl.extmodule for all firrtl.instance without its firrtl.module or firrtl.extmodule

sequencer, 2025-01-04:
This is for the separate linking flow, more detail description will be filed later.

uenoku, 2025-01-06:
I don't think we can support this. This is ill-formed IR and it's frontend's responsibility to produce well-formed IR. 

> Since our eDSL frontend only can operate on the firrtl.module block,: It's not possible to insert the firrtl.extmodule to firrtl.circuit at the time when adding the firrtl.instance to firrtl.module.

I'm not sure how your eDSL works but MLIR CAPI provides standard builder utilities so you should be able to change insertion point to firrtl.circuit body.

sequencer, 2025-01-06:
Thanks! I found it's simple to do the post-processing at the final stage of `firrtl.circuit` generation with the `mlirOperationWalk` API. Also discussed with @seldridge about this. I agree using a spacial pass to do so is not good at all. Since we should provide a valid IR for each stage.

closed 2025-01-06 by sequencer