llvm/circt #8050: [LoopScheduleToCalyx] Fail due to op still in use

KelvinChung2000, 2025-01-09:
When calling `circt-opt --lower-loopschedule-to-calyx ./test.mlir`,  `circt-opt` fail due to 
```
circt-opt: /home/kelvin/circt/llvm/mlir/lib/IR/PatternMatch.cpp:182: auto mlir::RewriterBase::eraseOp(mlir::Operation *)::(anonymous class)::operator()(mlir::Operation *) const: Assertion `mayBeGraphRegion(*op->getParentRegion()) && "expected that op has no uses"' failed.
```

test.mlir:
```mlir
module {
  func.func @foo(%arg0: i32) attributes {} {
    %const = arith.constant 1 : index
    loopschedule.pipeline II =  2 trip_count =  20 iter_args(%counter = %const) : (index) -> () {
      %latch = arith.cmpi ult, %counter, %const : index
      loopschedule.register %latch : i1
    } do {
      %stage0:2 = loopschedule.pipeline.stage start = 0 {
        %op = arith.addi %counter, %const : index
        %count = arith.addi %counter, %const : index
        loopschedule.register %op, %count : index, index
      } : index, index
      %stage1 = loopschedule.pipeline.stage start = 1 {
        loopschedule.register %stage0#0: index
      } : index
      %stage3 = loopschedule.pipeline.stage start = 7 {
        %moreOp = arith.addi %stage1, %const : index
        loopschedule.register %moreOp : index
      } : index
      loopschedule.terminator iter_args(%stage0#1), results() : (index) -> ()
    }
    return
  }
}
```

I have spent some time looking at what led to this, and it seems like the `%stage1` value is leading to the problem.

cgyurgyik, 2025-01-09:
Thanks Kelvin! @jiahanxie353 can you take a look at this please?

cgyurgyik, 2025-01-09:
Here is an even simpler repro:

```mlir
module {
  func.func @foo() attributes {} {
    %const = arith.constant 1 : index
    loopschedule.pipeline II =  2 trip_count =  20 iter_args(%counter = %const) : (index) -> () {
      %latch = arith.cmpi ult, %counter, %const : index
      loopschedule.register %latch : i1
    } do {
      %S0 = loopschedule.pipeline.stage start = 0 {
        %op = arith.addi %counter, %const : index
        loopschedule.register %op : index
      } : index
      %S1 = loopschedule.pipeline.stage start = 1 {
        loopschedule.register %S0: index
      } : index
      loopschedule.terminator iter_args(%S0), results() : (index) -> ()
    }
    return
  }
}
```

cgyurgyik, 2025-01-09:
This is probably related to the register being used in the while loop.

KelvinChung2000, 2025-01-09:
I am wondering if this would be an artificial problem because there should not be any reason to have any empty stage. Maybe only due to optimisation, which results in such a stage.

cgyurgyik, 2025-01-09:
Yeah either way, it shouldn't result in this perplexing crash.

closed 2025-01-09 by cgyurgyik

jiahanxie353, 2025-01-10:
> Thanks Kelvin! [@jiahanxie353](https://github.com/jiahanxie353) can you take a look at this please?

I'm currently in China and didn't have the chance to check the message due to time zone difference, sorry about that.
Thanks for fixing it, @cgyurgyik ! And thanks @KelvinChung2000 for raising the issue!

cgyurgyik, 2025-01-10:
No worries!

PR #8055 (2025-01-09): [LoopScheduleToCalyx] deduplicate groups within a ParOp.
Deduplicates multiple occurrences of the same Enable in a `ParOp`. This can be introduced automatically by a compiler lowering phase, so we add this as a canonicalization pattern. Additionally, ensures that the pipeline register (and not the pipeline stage) is used in Calyx computation assignments.

Fixes #8050.