llvm/circt #8126: [SCFToCalyx] Circular dependency between `init_instance` and `instance`'s execution

jiahanxie353, 2025-01-24:
Given a source program:
```mlir
module {
  func.func @func(%0 : i32) -> i32 {
    return %0 : i32
  }

  func.func @main() -> i32 {
    %0 = arith.constant 0 : i32
    %1 = func.call @func(%0) : (i32) -> i32 
    func.return %1 : i32
  } 
}
```
current lowering result is:
```mlir
module attributes {calyx.entrypoint = "main"} {
  calyx.component @func(%in0: i32, %clk: i1 {clk}, %reset: i1 {reset}, %go: i1 {go}) -> (%out0: i32, %done: i1 {done}) {
    %true = hw.constant true
    %ret_arg0_reg.in, %ret_arg0_reg.write_en, %ret_arg0_reg.clk, %ret_arg0_reg.reset, %ret_arg0_reg.out, %ret_arg0_reg.done = calyx.register @ret_arg0_reg : i32, i1, i1, i1, i32, i1
    calyx.wires {
      calyx.assign %out0 = %ret_arg0_reg.out : i32
      calyx.group @ret_assign_0 {
        calyx.assign %ret_arg0_reg.in = %in0 : i32
        calyx.assign %ret_arg0_reg.write_en = %true : i1
        calyx.group_done %ret_arg0_reg.done : i1
      }
    }
    calyx.control {
      calyx.seq {
        calyx.enable @ret_assign_0
      }
    }
  }
  calyx.component @main(%clk: i1 {clk}, %reset: i1 {reset}, %go: i1 {go}) -> (%out0: i32, %done: i1 {done}) {
    %true = hw.constant true
    %c0_i32 = hw.constant 0 : i32
    %ret_arg0_reg.in, %ret_arg0_reg.write_en, %ret_arg0_reg.clk, %ret_arg0_reg.reset, %ret_arg0_reg.out, %ret_arg0_reg.done = calyx.register @ret_arg0_reg : i32, i1, i1, i1, i32, i1
    %func_instance.in0, %func_instance.clk, %func_instance.reset, %func_instance.go, %func_instance.out0, %func_instance.done = calyx.instance @func_instance of @func : i32, i1, i1, i1, i32, i1
    calyx.wires {
      calyx.assign %out0 = %ret_arg0_reg.out : i32
      calyx.group @init_func_instance {
        calyx.assign %func_instance.reset = %true : i1
        calyx.assign %func_instance.go = %true : i1
        calyx.group_done %func_instance.done : i1
      }
      calyx.group @ret_assign_0 {
        calyx.assign %ret_arg0_reg.in = %func_instance.out0 : i32
        calyx.assign %ret_arg0_reg.write_en = %true : i1
        calyx.group_done %ret_arg0_reg.done : i1
      }
    }
    calyx.control {
      calyx.seq {
        calyx.seq {
          calyx.seq {
            calyx.enable @init_func_instance
            calyx.invoke @func_instance[](%func_instance.in0 = %c0_i32) -> (i32)
          }
          calyx.enable @ret_assign_0
        }
      }
    }
  } {toplevel}
}
```

I think the logic for:
```mlir
      calyx.group @init_func_instance {
        calyx.assign %func_instance.reset = %true : i1
        calyx.assign %func_instance.go = %true : i1
        calyx.group_done %func_instance.done : i1
      }
```
is problematic because it creates a circular dependency between the `done` signal of `init_func_instance` and `func_instance`:
1. since we have:
```mlir
          calyx.seq {
            calyx.enable @init_func_instance
            calyx.invoke @func_instance[](%func_instance.in0 = %c0_i32) -> (i32)
          }
```
`func_instance` won't be invoke until group `init_func_instance` is `done`;
2. since we have `calyx.group_done %func_instance.done : i1`, `init_func_instance` won't be `done` unless `func_instance` is `done`;
3. this circular dependency make the program stuck.

I'm thinking, is the fix simply just make `init_func_instance`'s `done` signal simply be `true`, just like its `reset`/`go` signals.

What do you think @rachitnigam @cgyurgyik @andrewb1999 ?

cgyurgyik, 2025-01-24:
To be honest, I'm not recalling why `init_func_instance` is even necessary.

jiahanxie353, 2025-01-24:
> `calyx.invoke` is syntactic sugar for creating a group that drives the cell being invoked. Shouldn't that be sufficient, i.e., can we just remove `init_func_instance` entirely, or am I forgetting some stale state issue?

you are right: https://github.com/calyxir/calyx/blob/8dec8c27b13ff8261f645a2fd2fcc9fea28e2017/calyx-opt/src/passes/compile_invoke.rs#L380

I'll resolve this issue by simply deleting the creation of all `init_xxx_instance`

closed 2025-01-27 by jiahanxie353