llvm/circt #8162: [Verif] `verif.symbolic_value` semantics are ambiguous

TaoBi22, 2025-01-31:
It looks to me like the semantics of the `verif.symbolic_value` are unclear right now - specifically, there's no information in the docs about how/when the value is allowed to change. We discussed it a little [here](https://github.com/llvm/circt/pull/7145#discussion_r1678878295) when the op was first added, but I'm not sure we really reached a conclusion before it was merged.

Ultimately the issue stems from the fact that there are two interpretations of what a symbolic value means. In the SMT solver world it means a value that can have one assignment, whereas from a hardware verification angle it could potentially refer to an input that can change value over time.

I had a chat with @maerhart and we think there are three options:

1. `symbolic_value` can have only one assignment. (This is my preferred option, as I think the name `symbolic_value` implies it only ever has one value. In this case I'd suggest we also add a `symbolic_input` op where the values can change)
2. `symbolic_value` can change value on a clock edge. I think this is what we assume in #7870 (and I believe is consistent with BTOR2), but the semantics are still pretty murky on this interpretation because we don't have an associated clock. If we pick this we think it the op needs a `!seq.clock` operand to determine when it can change.
3. `symbolic_value` can change value whenever. This wouldn't require any modification to the op, and I think it probably holds with the interpretation taken in #7870 since I guess the value would always be latched when it's not going into combinational ops, but I'm not sure I haven't missed a corner case to do with comparing the value in a property or assumption later on.

IMO any of these would be fine, I just think we need to clarify which is the case in the op description.

CC: @fabianschuiki @leonardt @dobios

dobios, 2025-01-31:
My original design of this was that it would directly mirror the `input` op in BTOR2 (which is why the original name of this op was `symbolic_input` before people decided that it was too confusing and should have a more generic name), the idea was to determine the clock they were associated to based on the clocked assertions they are used in, which is why I didn't include a clock in the original design. So the value changes every clock cycle.

I like the idea of distinguishing the two but I'm not sure how this would translate to any front-end as inputs are inferred as symbolic based on whether or not they are used in the definition of a formal test and never explicitly defined, again why I tried to keep the signature so simple. It's unclear to me if a newly distinguished set of symbolic would actually ever be used tbh

fabianschuiki, 2025-01-31:
Specifying this more precisely is a great idea! For what it's worth, [Yosys uses `(* ANYSEQ *)` and `(* ANYCONST *)` attributes on variables](https://symbiyosys.readthedocs.io/en/latest/verilog.html#unconstrained-variables) to mark them as having one constant symbolic value, or a new symbolic value at every timestep. If I recall correctly, the `ANYSEQ` value changes on every time step, not every clock cycle -- which might actually work for most cases, now that I think of it 🤔 

We could add a `verif.symbolic_constant` that mimics Yosys' `ANYCONST`, assuming one fixed value for all timesteps.

WDYT?

TaoBi22, 2025-01-31:
> the idea was to determine the clock they were associated to based on the clocked assertions they are used in, which is why I didn't include a clock in the original design. So the value changes every clock cycle.

That makes sense! I guess since they now appear outside clocked assertions (after LowerContracts) we need to be more explicit now though.

> `ANYSEQ` value changes on every time step, not every clock cycle -- which might actually work for most cases, now that I think of it 🤔

Yeah, I still haven't thought of a test case where that breaks. AFAICT the usage in LowerContracts shouldn't violate this assumption - I think the only case where this breaks anything is if you're comparing a value to itself measured at another point within a clock cycle, and I can't think of a way you could express that in the core dialects (thankfully we're free of nasty Verilog delays at this level!).

So it sounds to me like the consensus is to have it be able to vary on any timestep? Agreed that `verif.symbolic_constant` might be useful at some point, but since AFAIK we don't have a use for it currently maybe we wait til it's needed before we add it.

dobios, 2025-02-02:
> So it sounds to me like the consensus is to have it be able to vary on any timestep?

Yes that sounds like a good approach for now!

>Agreed that verif.symbolic_constant might be useful at some point, but since AFAIK we don't have a use for it currently maybe we wait til it's needed before we add it.

This could be an interesting addition, we just need to rethink how this would be expressed in firrtl land if we do that

TaoBi22, 2025-02-03:
Cool, sounds like we're agreed on it changing every timestep so #8177 adds that documentation.

I'm still personally slightly cautious of the naming, since it seems very counterintuitive to me that something called a `symbolic_value` can have multiple values. @dobios @fabianschuiki would you be opposed to renaming it to something like `symbolic_input`?

fabianschuiki, 2025-02-03:
I don't have strong feelings about the name 🙂. I remember @dobios originally proposed to call this `symbolic_input`, only to then have it change in opinionated squabbles... so this feels like vindication for that original proposal 😁

TaoBi22, 2025-02-03:
Ah, I now realise `symbolic_input` was the original name of the op! Sorry for missing that - if that conversation's already all been hashed out then probably not worth reviving it, I think the fact we have the semantics clarified in the docs now does the trick!

closed 2025-02-03 by TaoBi22