llvm/circt #8173: [ImportVerilog] Crash on ordering-methods-reverse test

fabianschuiki, 2025-02-02:
Running `circt-verilog` on https://chipsalliance.github.io/sv-tests-results/?v=circt_verilog+7.4.2+ordering-methods-reverse causes a crash.

```systemverilog
module top ();

string s[] = { "hello", "sad", "world" };

initial begin
	$display(":assert: (('%s' == 'hello') and ('%s' == 'sad') and ('%s' == 'world'))",
		s[0], s[1], s[2]);
	s.reverse;
	$display(":assert: (('%s' == 'world') and ('%s' == 'sad') and ('%s' == 'hello'))",
		s[0], s[1], s[2]);
end

endmodule
```

```
circt-verilog -I /root/sv-tests/sv-tests/tests/chapter-7/arrays/unpacked/ordering-methods --timescale=1ns/1ns --single-unit --top=top [tests/chapter-7/arrays/unpacked/ordering-methods/reverse.sv](https://chipsalliance.github.io/sv-tests-results/tests/chapter-7/arrays/unpacked/ordering-methods/reverse.sv.html)
[tests/chapter-7/arrays/unpacked/ordering-methods/reverse.sv:18:16](https://chipsalliance.github.io/sv-tests-results/tests/chapter-7/arrays/unpacked/ordering-methods/reverse.sv.html#l-18): error: expression of type '!moore.string' cannot be cast to a simple bit vector
string s[] = { "hello", "sad", "world" };
               ^
[tests/chapter-7/arrays/unpacked/ordering-methods/reverse.sv:18:25](https://chipsalliance.github.io/sv-tests-results/tests/chapter-7/arrays/unpacked/ordering-methods/reverse.sv.html#l-18): error: expression of type '!moore.string' cannot be cast to a simple bit vector
string s[] = { "hello", "sad", "world" };
                        ^
[tests/chapter-7/arrays/unpacked/ordering-methods/reverse.sv:18:32](https://chipsalliance.github.io/sv-tests-results/tests/chapter-7/arrays/unpacked/ordering-methods/reverse.sv.html#l-18): error: expression of type '!moore.string' cannot be cast to a simple bit vector
string s[] = { "hello", "sad", "world" };
                               ^
PLEASE submit a bug report to https://github.com/llvm/circt and include the crash backtrace.
Stack dump:
0.	Program arguments: circt-verilog -I /root/sv-tests/sv-tests/tests/chapter-7/arrays/unpacked/ordering-methods --timescale=1ns/1ns --single-unit --top=top [tests/chapter-7/arrays/unpacked/ordering-methods/reverse.sv](https://chipsalliance.github.io/sv-tests-results/tests/chapter-7/arrays/unpacked/ordering-methods/reverse.sv.html)
Stack dump without symbol names (ensure you have llvm-symbolizer in your PATH or set the environment var `LLVM_SYMBOLIZER_PATH` to point to it):
0  circt-verilog 0x000055976388cb47
1  circt-verilog 0x000055976388a9ee
2  circt-verilog 0x000055976388d21a
[...]
```

thomasnormal, 2026-03-03:
Rechecked on current `main` and I can no longer reproduce the reported crash / import failure.

I validated with the original shape:
```sv
module top ();
  string s[] = { "hello", "sad", "world" };
  initial begin
    $display("%s %s %s", s[0], s[1], s[2]);
    s.reverse;
    $display("%s %s %s", s[0], s[1], s[2]);
  end
endmodule
```
- `circt-verilog --timescale=1ns/1ns --single-unit --top=top ...` now succeeds
- `circt-sim` output shows expected reversed order (`world, sad, hello`)

To prevent regressions I added a semantic in-tree test in my fork:
- commit: `ea806f3b3`
- file: `test/Tools/circt-sim/string-dynamic-array-reverse-semantic.sv`
- validation: `build_test/bin/llvm-lit -sv test/Tools/circt-sim/string-dynamic-array-reverse-semantic.sv` (PASS)

If desired I can upstream this as a PR to close out #8173 with regression coverage.

thomasnormal, 2026-03-03:
I re-ran the repro from this issue on current tree and did not hit the crash.

Repro command used:
- 
- Exit status: 0 (no crash)

There is also now in-tree semantic coverage for this behavior:
- 
- Validates before/after string dynamic-array ordering across 

So this currently appears fixed on ; if you still see failures, please share the exact commit/flags and I can bisect.

thomasnormal, 2026-03-03:
Rechecked on current `main` in this workspace; I cannot reproduce the original failure mode.

Repro used (from issue body):
```sv
module top ();
string s[] = { "hello", "sad", "world" };
initial begin
  $display(":assert: (('%s' == 'hello') and ('%s' == 'sad') and ('%s' == 'world'))", s[0], s[1], s[2]);
  s.reverse;
  $display(":assert: (('%s' == 'world') and ('%s' == 'sad') and ('%s' == 'hello'))", s[0], s[1], s[2]);
end
endmodule
```

Commands:
- `build_test/bin/circt-verilog --ir-hw /tmp/issue8173.sv -o /tmp/issue8173_hw.mlir`
- `build_test/bin/circt-sim /tmp/issue8173_hw.mlir --top top --max-time=1000000000`

Observed:
- compile succeeds (no crash/type error)
- simulation prints the expected pre/post-reverse assertions.

There is also in-tree coverage for this shape:
- `test/Conversion/ImportVerilog/string-dynamic-array-reverse-no-crash.sv`
- `test/Tools/circt-sim/string-dynamic-array-reverse-semantic.sv`