llvm/circt #8219: [ESI]Assersion: dyn_cast on a non-existent value

hailongSun2000, 2025-02-11:
For example:
```
module {
  esi.manifest.sym @CoerceBundleTransform name "CoerceBundleTransform"
  hw.module @CoerceBundleTransform(in %b_in : !esi.bundle<[!esi.channel<i32> to "req", !esi.channel<i8> from "resp"]>,
                                   out b_out : !esi.bundle<[!esi.channel<i24> to "arg", !esi.channel<i16> from "result"]>)
                                   attributes {output_file = #hw.output_file<"CoerceBundleTransform.sv", includeReplicatedOps>} {

    %rawOutput, %valid = esi.unwrap.vr %result, %ready : i16
    %0 = comb.extract %rawOutput from 0 : (i16) -> i8
    %chanOutput, %ready = esi.wrap.vr %0, %valid : i8
    %req = esi.bundle.unpack %chanOutput from %b_in : !esi.bundle<[!esi.channel<i32> to "req", !esi.channel<i8> from "resp"]>
    %rawOutput_0, %valid_1 = esi.unwrap.vr %req, %ready_3 : i32
    %1 = comb.extract %rawOutput_0 from 0 : (i32) -> i24
    %chanOutput_2, %ready_3 = esi.wrap.vr %1, %valid_1 : i24
    %bundle, %result = esi.bundle.pack %chanOutput_2 : !esi.bundle<[!esi.channel<i24> to "arg", !esi.channel<i16> from "result"]>
    hw.output %bundle : !esi.bundle<[!esi.channel<i24> to "arg", !esi.channel<i16> from "result"]>
  }
}
```

`circt-opt --esi-clean-metadata --lower-esi-bundles --lower-esi-to-hw %s` will trigger a stack crash.

Please let me know if this command line is an error.

```
Casting.h:662: decltype(auto) llvm::dyn_cast(From *) [To = circt::esi::WrapValidReadyOp, From = mlir::Operation]: Assertion `detail::isPresent(Val) && "dyn_cast on a non-existent value"' failed.
PLEASE submit a bug report to https://github.com/llvm/circt and include the crash backtrace.
Stack dump:
0.	Program arguments: ./circt-opt /home/phoenix/test/esi.mlir --esi-clean-metadata --lower-esi-bundles --lower-esi-to-hw
 #0 0x0000638db276f9fd llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) /home/phoenix/work/circt/llvm/llvm/lib/Support/Unix/Signals.inc:796:11
 #1 0x0000638db276feeb PrintStackTraceSignalHandler(void*) /home/phoenix/work/circt/llvm/llvm/lib/Support/Unix/Signals.inc:872:1
 #2 0x0000638db276df56 llvm::sys::RunSignalHandlers() /home/phoenix/work/circt/llvm/llvm/lib/Support/Signals.cpp:105:5
 #3 0x0000638db27706a5 SignalHandler(int) /home/phoenix/work/circt/llvm/llvm/lib/Support/Unix/Signals.inc:413:1
 #4 0x00007e1453242520 (/lib/x86_64-linux-gnu/libc.so.6+0x42520)
 #5 0x00007e14532969fc __pthread_kill_implementation ./nptl/./nptl/pthread_kill.c:44:76
 #6 0x00007e14532969fc __pthread_kill_internal ./nptl/./nptl/pthread_kill.c:78:10
 #7 0x00007e14532969fc pthread_kill ./nptl/./nptl/pthread_kill.c:89:10
 #8 0x00007e1453242476 gsignal ./signal/../sysdeps/posix/raise.c:27:6
 #9 0x00007e14532287f3 abort ./stdlib/./stdlib/abort.c:81:7
#10 0x00007e145322871b _nl_load_domain ./intl/./intl/loadmsgcat.c:1177:9
#11 0x00007e1453239e96 (/lib/x86_64-linux-gnu/libc.so.6+0x39e96)
#12 0x0000638db2e1e988 decltype(auto) llvm::dyn_cast<circt::esi::WrapValidReadyOp, mlir::Operation>(mlir::Operation*) /home/phoenix/work/circt/llvm/llvm/include/llvm/Support/Casting.h:663:49
#13 0x0000638db2e82e83 (anonymous namespace)::RemoveWrapUnwrap::matchAndRewrite(mlir::Operation*, llvm::ArrayRef<mlir::Value>, mlir::ConversionPatternRewriter&) const /home/phoenix/work/circt/lib/Dialect/ESI/Passes/ESILowerToHW.cpp:190:14
#14 0x0000638db2e9220f mlir::ConversionPattern::matchAndRewrite(mlir::Operation*, llvm::ArrayRef<mlir::ValueRange>, mlir::ConversionPatternRewriter&) const /home/phoenix/work/circt/llvm/mlir/include/mlir/Transforms/DialectConversion.h:568:12
```

teqdruid, 2025-02-24:
Sorry @hailongSun2000, I've been on vacation and sprint mode. Regardless of whether or not the input and pass ordering is correct, the passes should never assert. I'll dig into this tomorrow.

hailongSun2000, 2025-02-26:
It's nothing. Emm... Yes, I think you're right 👍 , no matter what the order should be correct(never assert).

teqdruid, 2025-02-26:
Just had a chance to look at this. You have to run `--lower-esi-ports` before you lower to hw.

I'll address the assertion shortly.

hailongSun2000, 2025-02-26:
Yes! Haha. I've attempted that. Like this:
```
// RUN: circt-opt --esi-clean-metadata --lower-esi-bundles --lower-esi-ports --lower-esi-to-hw %s

module {
  hw.module @CoerceBundleTransform(in %b_in_req : i32, in %b_in_req_valid : i1, in %b_out_result : i16, in %b_out_result_valid : i1, in %b_in_resp_ready : i1, in %b_out_arg_ready : i1, out b_in_req_ready : i1, out b_out_result_ready : i1, out b_in_resp : i8, out b_in_resp_valid : i1, out b_out_arg : i24, out b_out_arg_valid : i1) attributes {output_file = #hw.output_file<"CoerceBundleTransform.sv", includeReplicatedOps>} {
    %0 = comb.extract %b_out_result from 0 : (i16) -> i8
    %1 = comb.extract %b_in_req from 0 : (i32) -> i24
    hw.output %b_out_arg_ready, %b_in_resp_ready, %0, %b_out_result_valid, %1, %b_in_req_valid : i1, i1, i8, i1, i24, i1
  }
}
```

teqdruid, 2025-02-26:
Thanks for the bug report!

commit 3f0aa622 (2025-02-26): [ESI] Fix assert on lower unwrap value from block arg

Fixes #8219
closed 2025-02-26 by teqdruid