llvm/circt #8226: [ImportVerilog] Problem with conditional + register

felixridoux, 2025-02-12:
Hi, I think I might have encountered a bug in the Verilog frontend of CIRCT. Below is the Verilog code in `bug.v`:

```verilog
module Bug (O, A, B, C);

input A, B, C;
output O;

reg O;

always @(posedge B)
  if (C)
    O = A;
  else
    O = 1'b1;

endmodule
```

When I run the following command:

```sh
circt-verilog -o bug.mlir bug.v
```

I get the following output in `bug.mlir`:

```mlir
module {
  hw.module @Bug(in %A : i1, in %B : i1, in %C : i1, out O : i1) {
    %0 = seq.to_clock %B
    %1 = seq.compreg %A, %0 : i1  
    hw.output %1 : i1
  }
}
```

It appears that `bug.v` and `bug.mlir` are not semantically equivalent. 

PS : I am a beginner with CIRCT, so it might just be a misuse from myself ^^

Best,

Félix

felixridoux, 2025-02-12:
Thanks a lot @maerhart ! I will have a look to your fix :)

seldridge, 2026-03-06:
@thomasnormal, @thomasahle: Due to current CIRCT project policy, we are unable to accept patches from the `thomasnormal/circt` fork.

For others: please do not upstream any patches from the `thomasnormal/circt` fork. Additionally, recognize that any comments made by the @thomasnormal account appear to be from one or more AI agents.

For more information, please see this Discourse post: https://discourse.llvm.org/t/circt-project-policy-on-contributions-from-thomasnormal-circt-fork/90075