llvm/circt #8231: Tests that require Z3 fail rather than being unsupported

mikeurbach, 2025-02-12:
I've been dealing with something in my circt integration tests for a while, and it has been an annoyance, but not a major issue. However, it actually masked a problem in https://github.com/llvm/circt/pull/8225. Basically, I haven't configured z3, and any test with `// REQUIRES: libz3` fails, rather than being unsupported. Does anyone else see this?

Separately, I need to just get z3 set up so I can run these tests during these LLVM bumps, but the current behavior seems wrong.

I didn't see any open issue about this, but please redirect me if this is a dupe.

For completeness, here is how I configure things:

```
cmake -B build -G Ninja llvm/llvm \
    -DCMAKE_BUILD_TYPE=Debug \
    -DCMAKE_C_COMPILER=clang \
    -DCMAKE_CXX_COMPILER=clang++ \
    -DCMAKE_EXPORT_COMPILE_COMMANDS=ON \
    -DLLVM_ENABLE_PROJECTS=mlir \
    -DLLVM_ENABLE_ASSERTIONS=ON \
    -DLLVM_EXTERNAL_PROJECTS=circt \
    -DLLVM_EXTERNAL_CIRCT_SOURCE_DIR=. \
    -DLLVM_TARGETS_TO_BUILD=host \
    -DLLVM_LIT_ARGS="-v --show-unsupported" \
    -DLLVM_USE_LINKER=gold \
    -DMLIR_ENABLE_BINDINGS_PYTHON=ON \
    -DCIRCT_BINDINGS_PYTHON_ENABLED=ON
```

I will note that I see this in the output:

```
CMake Warning at cmake/modules/FindZ3.cmake:116 (message):
  Failed to determine Z3 library version, defaulting to 0.0.0.
Call Stack (most recent call first):
  /scratch/verif/mikeu/circt/CMakeLists.txt:474 (find_package)

-- Could NOT find Z3 (missing: Z3_LIBRARIES Z3_INCLUDE_DIR) (found version "0.0.0")
-- Did NOT find z3.
```

teqdruid, 2025-02-13:
> You are receiving this email because you are subscribed to this thread.

I get that message at config time but I don't get a test failure as a
result.

TaoBi22, 2025-03-20:
Took a poke around and think I've figured out what the problem is here, will get a PR in shortly

mikeurbach, 2025-03-21:
Confirmed the fix does the trick; on my test machine without Z3, these tests are now correctly marked as unsupported. Thanks @TaoBi22 .

closed 2025-03-21 by mikeurbach