llvm/circt #8268: Support dynamic indexed vector element in Target

unlsycn, 2025-02-23:
CIRCT currently cannot resolve annotations referencing dynamically indexed (non-literal) vector elements, such as:
```firrtl
  {
    "class":"firrtl.transforms.DontTouchAnnotation",
    "target":"~Foo|Foo>in[idx].foo"
  }
```
A semantic concern arises regarding whether the index expression [idx] introduces ambiguity, which this could potentially reference different signals, or we can just confines the search to it within the same module as the target vector.

seldridge, 2025-02-24:
I don't want to support this. Can you provide more information about your use case?

Looking at https://github.com/chipsalliance/chisel/issues/4575#issue-2757104306, it seems like you want to use the trace feature to track the name of some signal that is provided a dynamic index. Instead, can you trace all signals? I.e., trace the vector itself or trace all the leaves? In the example, there's very little difference between `~Foo|Foo>in[idx].foo` and `~Foo|Foo>in[*].foo`. While the latter could be supported, I'd opt to just support it via multiple annotations like `~Foo|Foo>in[0].foo`, `~Foo|Foo>in[1].foo`, ..., `~Foo|Foo>in[n-1].foo`. Would the multiple annotation approach work for you? I expect Chisel could be changed to emit it like this.

I do think what you are asking for is possible, though it would be very complicated to support in the compiler, to the point that I don't think it would be useful. This is saying that there is an annotation that points to different things based on a runtime value. The runtime value, shown above is a value inside the current scope. However, I assume it could be anything and this would actually have to be another annotation for the index.

unlsycn, 2025-04-10:
Hmmm... When I reconsider this issue, I realize that it is more convenient to do this expansion in CIRCT compared to Chisel, since in Chisel we cannot tell if a target contains a dynamic indexed element in its path before we actually resolve it, which then we are hard to know whether we should expand it in the `toTarget` process. And we would better not expand all annotations with dynamic target to multiple annotations or substitute the dynamic index with `*`, as certain annotations may need to preserve the original target.

closed 2025-04-12 by unlsycn