llvm/circt #8283: [ImportVerilog] Cannot compile forward decleared string type

Max-astro, 2025-02-28:
I want to support this [sv-tests-results/string_concat_op](https://chipsalliance.github.io/sv-tests-results/?v=circt_verilog+11.4.12.2+string_concat_op) feature, but not sure how to fix it properly.

``` sv
module top();

string str;

initial begin
    str = "Hello";
end

endmodule
```

`circt-verilog`  has the following complaints when compiling the string.sv file:
```s.sv:3:8: error: failed to legalize operation 'moore.variable'
string str;
       ^
s.sv:3:8: note: see current operation: %0 = "moore.variable"() <{name = "str"}> : () -> !moore.ref<string>
```

This error is due to MooreToCore's lack of string-type conversion. It only has `StringConstantOpConv` and `Format*OpConversion`; none of these conversions can handle the `string` type.

chenbo-again, 2025-03-03:
I found hw dialect has HWStringType, and maybe it's a proper target for lowering moore.string.

fabianschuiki, 2025-03-03:
Thanks for working on this, @Max-astro! HW doesn't have very good string support -- it's limited mostly to fixed-sized strings or strings that can be passed to module instances as parameter. There has been a recent effort in the Sim dialect, which collects simulation-related things, to come up with ways to do string formatting and representing format string literals. I think the Sim dialect would be a great place to add a type for dynamic strings such as Verilog's `string` type, and also have operations like concatenation, casting, and indexing. Maybe that would be a way forward for you?

Max-astro, 2025-03-04:
> I think the Sim dialect would be a great place to add a type for dynamic strings such as Verilog's `string` type, and also have operations like concatenation, casting, and indexing.

Thank you for your advice, I'll take a look at the Sim dialect and see if it's easy to add a dynamic string type.

thomasnormal, 2026-03-03:
Rechecked on current `main`; this issue appears fixed.

Repro from issue now succeeds:
```sv
module top();
  string str;
  initial str = "Hello";
endmodule
```

`circt-verilog` no longer errors on `moore.variable` string lowering for this case.

Unless there is another failing string variant, this looks closeable as stale/fixed.