llvm/circt #8286: [circt-verilog][llhd][arcilator] Verilog-to-LLVM lowering issues

likeamahoney, 2025-03-03:
Hi all!

There is few issues related to moore to llvm lowering pipeline.

Currently there is no possibility to lower combination logic with control flow operators into LLVM. For example:

```verilog
module  top  (
    input           clk ,
    input           rstn ,
    input      in
    output     out
    );

    parameter            P1   = 3'd0 ;
    parameter            P2  = 3'd3 ;

    reg [2:0]            r1 = 3'd1;

    always_comb begin
        if (in)
            r1 = P1;
        else
            r1 = P2;
    end
endmodule
```

because it fails with this : `error: failed to legalize operation 'llhd.constant_time'`

And also `arcilator` fails to lower sequential logic which contains both blocking and nonblocking assignments like this example:

```verilog
module  top  (
    input           clk ,
    input           rstn ,
    input      in
    output     out
    );

    parameter            P1   = 3'd0 ;
    parameter            P2  = 3'd3 ;

    reg [2:0]            r1 = 3'd1;

    always @(posedge clk or negedge rstn) begin
        r1 = P1;
        r1 <= P2 + r1;
    end

endmodule
```

because it fails with that:

```
 error: body contains non-pure operation
    %2 = comb.add %1, %c3_i3 : i3
```

I'm using to lower a such cmd:

```
circt-verilog sample.sv | arcilator 
```

Also I tried to add into "populate LLHD" pipeline a newly added passes  -- "llhd-hoist-signals" and "llhd-mem2reg". It didn't help.

1) What are the plans for improving verilog-to-llvm lowering  and do they exist at all?
2) Will the "llhd-hoist-signals" and "llhd-mem2reg" passes be added to the `circt-verilog` tool or will they stay standalone?

@fabianschuiki @maerhart

fabianschuiki, 2025-03-03:
Thanks for the fantastic issue report @likeamahoney! 🥳 The HoistSignals and Mem2Reg passes are an effort to make the Verilog lowering to HW and Arcilator/LLVM more reliable. The idea is to run Mem2Reg and then HoistSignals to simplify processes, then run an improved version of Desequentialize to detect registers/latches and factor them out, and then convert the LLHD processes to `scf.execute_region`s. At that point Arcilator should be able to simulate basic synthesizable structures. A bit later, we'll want to add support for the LLHD dialect to Arcilator, such that it can simulate the progression of physical time and maintain an event queue where it's needed.

So the idea is definitely to get this into a shape where it can deal with any Verilog you throw at it 😁

likeamahoney, 2025-03-04:
> Thanks for the fantastic issue report [@likeamahoney](https://github.com/likeamahoney)! 🥳 The HoistSignals and Mem2Reg passes are an effort to make the Verilog lowering to HW and Arcilator/LLVM more [...]

Thank you very much for writing a detailed roadmap. I really hope that soon it will be possible to simulate large `SystemVerilog` designs on the `arcilator`.