llvm/circt #8375: Modeling inverter and buffer in the core dialect

Max-astro, 2025-04-01:
I'm currently attempting to model the gate-level netlist using the core dialect; however, it lacks operations for buffers and inverters, which are commonly used in circuit designs.

I'm wondering if I can add these two operations into the `comb` dialect, or if I should create a new dialect to model the gate-level netlist, regardless of whether it's technology-dependent or not.

TaoBi22, 2025-04-01:
Hi @Max-astro,

The `comb` dialect deliberately doesn't provide an inverter op to cut down on unnecessary ops, since there are other ways to represent this op (I believe XORing with a constant -1 is what we usually use in CIRCT) - if you'd like some more info on that, there's a paragraph about it in the comb dialect rationale [here](https://circt.llvm.org/docs/Dialects/Comb/RationaleComb/#no-complement-negate-zext-sext-operators).

Buffers aren't something I would consider combinational, which is what the `comb` dialect is intended to contain. I'm not sure what kind of buffer you're looking for, but it's possible the [FIFO operation in the `seq` dialect](https://circt.llvm.org/docs/Dialects/Seq/RationaleSeq/#the-fifo-operation) is what you're looking for. The `seq` dialect is where something like a buffer would normally live, unless it's some kind of higher-level abstraction over buffers.

TaoBi22, 2025-04-01:
That makes sense! Sorry, I somehow missed the motivation part of the issue...

It sounds to me like, as you suggested, another dialect might be easiest for this - that way you can represent what you need explicitly without messing with the canonicalizations etc. that motivate not having things like inversion in comb.

Max-astro, 2025-04-01:
> It sounds to me like, as you suggested, another dialect might be easiest for this - that way you can represent what you need explicitly without messing with the canonicalizations etc. that motivate not having things like inversion in comb.

Thank you for your valuable information and suggestions. 

I plan to introduce a new dialect to model common generic gate primitives. 
With the existing core dialect and this new addition, we should be able to comprehensively describe a technology-independent gate-level netlist. 

Additionally, I,m concerned about submitting a pull request to create a new dialect. Will it be difficult to gain acceptance from the community?

closed 2025-04-02 by Max-astro

TaoBi22, 2025-04-02:
> Additionally, I,m concerned about submitting a pull request to create a new dialect. Will it be difficult to gain acceptance from the community?

Getting a new dialect in might have a slightly higher bar than the average PR, but you have a clear use case for it and I think it's an area the wider community is interested in improving support for!