llvm/circt #8411: [circt-bmc] Crash with `--print-solver-output`

bryceberger, 2025-04-11:
This appears to be due to the wrong signature for `printf` generated by `lib/Tools/circt-bmc/LowerToBMC.cpp` (`(ptr, ...) -> void` instead of `(ptr, ...) -> i32`).

Backtrace:

```
file.mlir:1:1: error: redefinition of function 'printf' of different type '!llvm.func<i32 (ptr, ...)>' is prohibited
module {
^
circt-bmc: /home/$USER/repos/github/llvm/circt/lib/Conversion/SMTToZ3LLVM/LowerSMTToZ3LLVM.cpp:155: mlir::LLVM::CallOp {anonymous}::SMTLoweringPattern<OpTy>::buildCall(mlir::OpBuilder&, mlir::Location, llvm::StringRef, mlir::LLVM::LLVMFunctionType, mlir::ValueRange) const [with OpTy = circt::smt::CheckOp]: Assertion `succeeded(funcOpResult) && "expected to lookup or create printf"' failed.
PLEASE submit a bug report to https://github.com/llvm/circt and include the crash backtrace.
Stack dump:
0.	Program arguments: circt-bmc --shared-libs /lib/x86_64-linux-gnu/libz3.so.4 --print-solver-output --module ex file.mlir -b 1
 #0 0x000064b034d68df0 llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) (/home/$USER/repos/github/llvm/circt/build/bin/circt-bmc+0x193edf0)
 #1 0x000064b034d6627f llvm::sys::RunSignalHandlers() (/home/$USER/repos/github/llvm/circt/build/bin/circt-bmc+0x193c27f)
 #2 0x000064b034d663ca SignalHandler(int, siginfo_t*, void*) Signals.cpp:0:0
 #3 0x0000793448442520 (/lib/x86_64-linux-gnu/libc.so.6+0x42520)
 #4 0x00007934484969fc __pthread_kill_implementation ./nptl/pthread_kill.c:44:76
 #5 0x00007934484969fc __pthread_kill_internal ./nptl/pthread_kill.c:78:10
 #6 0x00007934484969fc pthread_kill ./nptl/pthread_kill.c:89:10
 #7 0x0000793448442476 gsignal ./signal/../sysdeps/posix/raise.c:27:6
 #8 0x00007934484287f3 abort ./stdlib/abort.c:81:7
 #9 0x000079344842871b _nl_load_domain ./intl/loadmsgcat.c:1177:9
#10 0x0000793448439e96 (/lib/x86_64-linux-gnu/libc.so.6+0x39e96)
#11 0x000064b035639692 (anonymous namespace)::SMTLoweringPattern<circt::smt::CheckOp>::buildCall(mlir::OpBuilder&, mlir::Location, llvm::StringRef, mlir::LLVM::LLVMFunctionType, mlir::ValueRange) const LowerSMTToZ3LLVM.cpp:0:0
#12 0x000064b0356303e1 (anonymous namespace)::CheckOpLowering::matchAndRewrite(circt::smt::CheckOp, circt::smt::CheckOpAdaptor, mlir::ConversionPatternRewriter&) const LowerSMTToZ3LLVM.cpp:0:0
#13 0x000064b0356a3f40 mlir::OpConversionPattern<circt::smt::CheckOp>::matchAndRewrite(circt::smt::CheckOp, circt::smt::CheckOpGenericAdaptor<llvm::ArrayRef<mlir::ValueRange>>, mlir::ConversionPatternRewriter&) const LowerSMTToZ3LLVM.cpp:0:0
#14 0x000064b0356a3c90 mlir::OpConversionPattern<circt::smt::CheckOp>::matchAndRewrite(mlir::Operation*, llvm::ArrayRef<mlir::ValueRange>, mlir::ConversionPatternRewriter&) const LowerSMTToZ3LLVM.cpp:0:0
#15 0x000064b0374e33cc mlir::ConversionPattern::matchAndRewrite(mlir::Operation*, mlir::PatternRewriter&) const (/home/$USER/repos/github/llvm/circt/build/bin/circt-bmc+0x40b93cc)
#16 0x000064b037514208 mlir::PatternApplicator::matchAndRewrite(mlir::Operation*, mlir::PatternRewriter&, llvm::function_ref<bool (mlir::Pattern const&)>, llvm::function_ref<void (mlir::Pattern const&)>, llvm::function_ref<llvm::LogicalResult (mlir::Pattern const&)>) (/home/$USER/repos/github/llvm/circt/build/bin/circt-bmc+0x40ea208)
#17 0x000064b0374e5cfc (anonymous namespace)::OperationLegalizer::legalize(mlir::Operation*, mlir::ConversionPatternRewriter&) DialectConversion.cpp:0:0
#18 0x000064b0374e627c mlir::OperationConverter::convert(mlir::ConversionPatternRewriter&, mlir::Operation*) (/home/$USER/repos/github/llvm/circt/build/bin/circt-bmc+0x40bc27c)
#19 0x000064b0374e9a4c mlir::OperationConverter::convertOperations(llvm::ArrayRef<mlir::Operation*>) (/home/$USER/repos/github/llvm/circt/build/bin/circt-bmc+0x40bfa4c)
#20 0x000064b0374eb109 mlir::applyFullConversion(mlir::Operation*, mlir::ConversionTarget const&, mlir::FrozenRewritePatternSet const&, mlir::ConversionConfig) (/home/$USER/repos/github/llvm/circt/build/bin/circt-bmc+0x40c1109)
#21 0x000064b03563369f (anonymous namespace)::LowerSMTToZ3LLVMPass::runOnOperation() LowerSMTToZ3LLVM.cpp:0:0
#22 0x000064b03757f5b1 mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) (/home/$USER/repos/github/llvm/circt/build/bin/circt-bmc+0x41555b1)
#23 0x000064b03757fa61 mlir::detail::OpToOpPassAdaptor::runPipeline(mlir::OpPassManager&, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) (/home/$USER/repos/github/llvm/circt/build/bin/circt-bmc+0x4155a61)
#24 0x000064b037580b55 mlir::PassManager::run(mlir::Operation*) (/home/$USER/repos/github/llvm/circt/build/bin/circt-bmc+0x4156b55)
#25 0x000064b034cbe6b4 executeBMC(mlir::MLIRContext&) circt-bmc.cpp:0:0
#26 0x000064b034cbf3a3 main (/home/$USER/repos/github/llvm/circt/build/bin/circt-bmc+0x18953a3)
#27 0x0000793448429d90 __libc_start_call_main ./csu/../sysdeps/nptl/libc_start_call_main.h:58:16
#28 0x0000793448429e40 call_init ./csu/../csu/libc-start.c:128:20
#29 0x0000793448429e40 __libc_start_main ./csu/../csu/libc-start.c:379:5
#30 0x000064b034cb8c65 _start (/home/$USER/repos/github/llvm/circt/build/bin/circt-bmc+0x188ec65)
fish: Process 164480, 'circt-bmc' from job 1, 'circt-bmc --shared-libs /lib/x8…' terminated by signal SIGABRT (Abort)
```

TaoBi22, 2025-04-14:
Hmm this is interesting. This seems to have been caused by #8327, which fixed the same bug except that it wanted a void type rather than an i32 type. In fact, main works on my machine but the branch with your proposed fix crashes. Clearly there's some kind of environment inconsistency here, since our machines seem to disagree about what is acceptable.

Since your PR crashes on my machine, it's probably not the fix we want - nonetheless, thanks for looking into it and opening a PR! There is an alternative way to do this using a function helper that we planned to transition to anyway, so I will open a PR doing it that way and ping you so we can make sure that one works on both our machines before we merge it :)

bryceberger, 2025-04-14:
Hmm, strange. For the record, I compiled both circt and llvm with `-DCMAKE_BUILD_TYPE=Release`.

TaoBi22, 2025-04-14:
Yeah it's release mode on my end too, though I'm guessing it's more likely to be something lib related than build modes.

What's in your input file for the output you posted above, is it just logic?

bryceberger, 2025-04-14:
Yeah, just a single output.

```mlir
module {
  hw.module @ex(in %i: i1, out o: i1) {
    verif.assert %i label "" : i1
    hw.output %i : i1
  }
}
```

bryceberger, 2025-04-14:
https://github.com/llvm/circt/pull/8412#issuecomment-2802268016

Seems to have been a submodule error on my part.

closed 2025-04-14 by bryceberger