llvm/circt #8441: [MLIR Complation]: Failed  when covert MLIR to Verilog in some cases.

Siudya, 2025-04-24:
I use firtool 1.62.1 to compile 2 similar mlir files, but one failed.

[bad.hw.mlir.gz](https://github.com/user-attachments/files/19889211/bad.hw.mlir.gz)

[good.hw.mlir.gz](https://github.com/user-attachments/files/19889212/good.hw.mlir.gz)

```bash
gzip -d good.hw.mlir.gz
firtool --format=mlir -o good.sv --verilog good.hw.mlir # success in about 10 seconds
```

```bash
gzip -d bad.hw.mlir.gz
firtool --format=mlir -o bad.sv --verilog bad.hw.mlir # failed in about 2 hours
```

seldridge, 2025-04-24:
Try running with `-verbose-pass-executions` to see where it is getting stuck for that version. I ran this on a 1.61.0 binary I had downloaded and it is getting stuck (or taking a while) on `lower-seq-to-sv`. This pass has had known performance issues on some versions which were fixed with a new algorithm and then a secondary fix recently which caps the total number of if/else statements that can be generated.

I expect that this is definitely fixed on firtool 1.112.0, but may be fixed on firtool 1.73.0.

If you have the input .fir files that would be easier to test with as the .mlir format has changed.

Siudya, 2025-04-25:
[small_good.hw.mlir.gz](https://github.com/user-attachments/files/19902485/small_good.hw.mlir.gz)

[small_bad.hw.mlir.gz](https://github.com/user-attachments/files/19902486/small_bad.hw.mlir.gz)

I uploaded 2 smaller mlir files.

```bash
gzip -d good.hw.mlir.gz
firtool --format=mlir -o good.sv --verilog good.hw.mlir # complete in about 30 seconds
```

```bash
gzip -d bad.hw.mlir.gz
firtool --format=mlir -o bad.sv --verilog bad.hw.mlir # complete in about 20 minutes
```

small_bad.hw.mlir and small_good.hw.mlir are generated from the same buffer-like module, but small_bad.hw.mlir has 4 entries and small_good.hw.mlir has 2. Compile time of small_bad is much longer. 30s of compile time of small_good is also abnormal by the way.

I use Chisel 6.7.0 for my design, witch uses firtool 1.62.1. Issue#7586 fix a bug in 1.62.0 that 1.75.0 also has. I don't which firtool I should use now.

Siudya, 2025-04-25:
[small_bad.fir.gz](https://github.com/user-attachments/files/19902716/small_bad.fir.gz)

[small_good.fir.gz](https://github.com/user-attachments/files/19902717/small_good.fir.gz)

I uploaded 2 smaller .fir files.

```bash
gzip -d small_good.fir.gz
firtool small_good.fir --format=fir -o small_good.sv --disable-annotation-unknown --verilog # complete in about 40 seconds
```

```bash
gzip -d small_bad.fir.gz
firtool small_bad.fir --format=fir -o small_bad.sv --disable-annotation-unknown --verilog # complete in about 20 minutes
```

`small_bad.fir` and `small_good.fir` are generated from the same buffer-like module, but `small_bad` has 4 entries while `small_good` has 2. Compile time of `small_bad` is much longer. 40s of compile time of small_good is also abnormal by the way.

Both 1.75.0 and 1.62.1 had been tried but 1.75.0 even took 20% more time to compile. 

I use Chisel 6.7.0 for my design, which uses firtool 1.62.1. Which firtool I should use?.

Siudya, 2025-04-27:
[small_bad.fir.gz](https://github.com/user-attachments/files/19902716/small_bad.fir.gz)

[small_good.fir.gz](https://github.com/user-attachments/files/19902717/small_good.fir.gz)

I uploaded 2 smaller .fir files.

```bash
gzip -d small_good.fir.gz
firtool small_good.fir --format=fir -o small_good.sv --disable-annotation-unknown --verilog # complete in about 40 seconds
```

```bash
gzip -d small_bad.fir.gz
firtool small_bad.fir --format=fir -o small_bad.sv --disable-annotation-unknown --verilog # complete in about 20 minutes
```

`small_bad.fir` and `small_good.fir` are generated from the same buffer-like module, but `small_bad` has 4 entries while `small_good` has 2. Compile time of `small_bad` is much longer. 40s of compile time of small_good is also abnormal by the way.

Both 1.75.0 and 1.62.1 had been tried but 1.75.0 performed even worse (50 seconds on `small_good` and not complete in 2 hours on `small_bad`). 

I use Chisel 6.7.0 for my design, the latest chisel release for now, which still uses firtool 1.62.1. An incompatible firtool is not an option for me.

closed 2025-06-10 by Siudya