llvm/circt #8476: [MooreToCore] Lower exponentiation to `math.ipowi`

fabianschuiki, 2025-05-12:
@jiahanxie353 has added the Math dialect to circt-opt in #8475. This is a great precedent to start using that dialect in the MooreToCore conversion. For example, the `moore.pow` operation could be lowered to `math.ipowi` and `math.powf` depending on its type.

fabianschuiki, 2025-06-05:
Hey @liamslj13, great to have you on board! And don't worry, the codebase definitely takes a while to get used to. Once you've seen and edited a few files in different corners it becomes a lot easier 😃. The `MooreToCore` pass is a conversion, which lives in `Conversion/MooreToCore` instead of the usual dialect passes like `Dialect/Moore/Transforms`:

https://github.com/llvm/circt/blob/main/lib/Conversion/MooreToCore/MooreToCore.cpp

This file contains all the conversion patterns from the Moore dialect to HW/Comb/Seq. I think there are even existing ones for unsigned and signed "pow" operations in the Moore dialect. It would be really nice if those could call into `math.ipowi` instead 😃

liamslj13, 2025-06-06:
Hi again, I've been looking through the file that we need to change and would just like to clarify a few things that I am still a bit confused about.
```cpp
struct PowUOpConversion : public OpConversionPattern<PowUOp> {
  using OpConversionPattern::OpConversionPattern;

  LogicalResult
  matchAndRewrite(PowUOp op, OpAdaptor adaptor,
                  ConversionPatternRewriter &rewriter) const override {
    Type resultType = typeConverter->convertType(op.getResult().getType());

    Location loc = op.getLoc();
    auto intType = cast<IntType>(op.getRhs().getType());

    // transform a ** b into scf.for 0 to b step 1 { init *= a }, init = 1
    Type integerType = rewriter.getIntegerType(intType.getWidth());
    Value lowerBound = rewriter.create<hw::ConstantOp>(loc, integerType, 0);
    Value upperBound =
        rewriter.create<ConversionOp>(loc, integerType, op.getRhs());
    Value step = rewriter.create<hw::ConstantOp>(loc, integerType, 1);

    Value initVal = rewriter.create<hw::ConstantOp>(loc, resultType, 1);
    Value lhsVal = rewriter.create<ConversionOp>(loc, resultType, op.getLhs());

    auto forOp = rewriter.create<scf::ForOp>(
        loc, lowerBound, upperBound, step, ValueRange(initVal),
        [&](OpBuilder &builder, Location loc, Value i, ValueRange iterArgs) {
          Value loopVar = iterArgs.front();
          Value mul = rewriter.create<comb::MulOp>(loc, lhsVal, loopVar);
          rewriter.create<scf::YieldOp>(loc, ValueRange(mul));
        });

    rewriter.replaceOp(op, forOp.getResult(0));

    return success();
  }
};
```
I believe it is this function and the subsequent **PowSOpConversion** that we need to edit. With regards to this code, I instead should directly call **ipowi**op from the Math dialect that was integrated into MLIR in the PR linked above, and this logic would replace the for loop at the bottom that performs the IR conversion for exponentiation?

Apologies for being so slow to understand this.

fabianschuiki, 2025-06-09:
Yes that sounds like a great plan! You may have to take a close look at the documentation for `ipowi` to figure out whether the exponent is signed or unsigned, and then think about how the two Moore dialect operations map to it 😃

liamslj13, 2025-06-10:
I may have made some progress, but I am not sure I am doing it correctly. I have reviewed relevant documentation as well.
```cpp
struct PowSOpConversion : public OpConversionPattern<PowSOp> {
  using OpConversionPattern::OpConversionPattern;

  LogicalResult
  matchAndRewrite(PowSOp op, OpAdaptor adaptor,
                  ConversionPatternRewriter &rewriter) const override {
    Type resultType = typeConverter->convertType(op.getResult().getType());

    Location loc = op.getLoc();
    auto intType = cast<IntType>(op.getRhs().getType());

    Type integerType = rewriter.getIntegerType(intType.getWidth());
    Value lhsVal = rewriter.create<ConversionOp>(loc, resultType, op.getLhs());
    Value rhsVal = rewriter.create<ConversionOp>(loc, integerType, op.getRhs());
    // utilize MLIR math dialect's math.ipowi to handle the exponentiation of expression
    Value result = rewriter.create<mlir::math::IPowIOp>(loc, lhsVal, rhsVal);

    rewriter.replaceOp(op, result);

    return success();
  }
};
```
The for loop has been removed and I think I have replaced it with the relevant syntax to lower it to MLIR. I also added the relevant `#include "mlir/Dialect/Math/IR/Math.h"` at the top. Please let me know if there are any mistakes 🙇

liamslj13, 2025-06-12:
Apologies for the delayed response! I have removed these four lines,
```cpp
auto intType = cast<IntType>(op.getRhs().getType());
Type integerType = rewriter.getIntegerType(intType.getWidth());
Value lhsVal = rewriter.create<ConversionOp>(loc, resultType, op.getLhs());
Value rhsVal = rewriter.create<ConversionOp>(loc, integerType, op.getRhs());
```
and left what you have suggested.
```cpp
struct PowSOpConversion : public OpConversionPattern<PowSOp> {
  using OpConversionPattern::OpConversionPattern;

  LogicalResult
  matchAndRewrite(PowSOp op, OpAdaptor adaptor,
                  ConversionPatternRewriter &rewriter) const override {
    Type resultType = typeConverter->convertType(op.getResult().getType());

    Location loc = op.getLoc();
    // utilize MLIR math dialect's math.ipowi to handle the exponentiation of expression
    Value result = rewriter.create<mlir::math::IPowIOp>(loc, adaptor.getLhs(), adaptor.getRhs());

    rewriter.replaceOp(op, result);

    return success();
  }
};
```
As for the types of `math::IPowIOp` types, it says in the documentation `signless-integer-like` for both the `LHS` and `RHS`, as well as `result` fields. Should I try to write a test case somewhere to ensure this doesn't fail when doing `check-circt`? I also need to figure out how to zero/sign extend if needed.

liamslj13, 2025-06-17:
Awesome! I will open a PR right now for the PowSOpConversion branch first and see if we can merge or make any further changes. Thank you for all of the guidance thus far 🙌