llvm/circt #8485: `seq.FIFO` fails at `lower-seq-to-sv` in PyCDE

were, 2025-05-16:
Dear All,

I am getting familiar with circt and I am trying to write something simple:

````Python
class Top(Module):
    clk = Clock()
    rst = Reset()
    a = Input(types.i32)
    exposed_c = Output(types.i32)

    @generator
    def construct(self):
        fifo = FIFO(type=types.i32, depth=4, clk=self.clk, rst=self.rst, rd_latency=1)
        one = types.i1(1)
        fifo.push(self.a, one)
        fifo.pop(one)
        self.exposed_c = fifo.output
````

It gives me the error log below:
````
failed to legalize operation 'seq.fifo' that was explicitly marked illegal
see current operation: %1:5 = "seq.fifo"(<<UNKNOWN SSA VALUE>>, %0, %0, <<UNKNOWN SSA VALUE>>, <<UNKNOWN SSA VALUE>>) <{depth = 4 : i64, rdLatency = 1 : i64, resultSegmentSizes = array<i32: 1, 1, 1, 1, 1>}> : (i32, i1, i1, !seq.clock, i1) -> (i32, i1, i1, i1, i1)
````

It turns out that those `<<UNKNOWN SSA VALUE>>` are `self.a`, `self.clk`, and `self.rst`, which indicates a bad hierarchical synthesis.

teqdruid, 2025-05-16:
Thanks for the bug report! Try it now.

teqdruid, 2025-05-16:
Also, I'm trying to get rid of the `types` class. `Bits(32)` is the preferred way to get an i32 now.

closed 2025-05-16 by teqdruid

were, 2025-05-17:
> Also, I'm trying to get rid of the `types` class. `Bits(32)` is the preferred way to get an i32 now.

@teqdruid OK. I will avoid further usage of that. I just copy-pasted that from the test cases.

teqdruid, 2025-05-17:
Yeah, I haven't updated all of the test cases yet.

commit 45e61df2 (2025-05-16): [PyCDE] Fix FIFO lowering bug #8485 be adding missing passes

The FIFO lowering flow was missing several passes. Fixes #8485.