llvm/circt #8514: [Comb] Missed optimization for `icmp(extract, c)`

uenoku, 2025-05-23:
```mlir
hw.module @test(in %a : i4, out b : i1) {
   %r = comb.extract %a from 1 : (i4) -> i3
   %c_1_i3 = hw.constant 1 : i3
   %icmp = comb.icmp eq %r, %c_1_i3 : i3
   hw.output %icmp : i1
}
```

```verilog
$ circt-opt bar.mlir -canonicalize -export-verilog
// Generated by CIRCT unknown git version
module test(    // bar.mlir:4:1
  input  [3:0] a,       // bar.mlir:4:20
  output       b        // bar.mlir:4:33
);

  assign b = a[3:1] == 3'h1;    // bar.mlir:5:9, :7:12, :8:4
endmodule
```

this could be `assign b = a[1]`;

dtzSiFive, 2025-05-23:
This doesn't seem equivalent, `b = a[1]` doesn't check the top two bits of `a` are zero like the original?

uenoku, 2025-05-23:
Yeah good point, thanks.`a` was one-hot encoded in original code so it was valid, but yes it's illegal in general.

closed 2025-05-23 by uenoku