llvm/circt #8519: [firtool] Duplicate include directives for `layers-*.sv` files after dedup

tymcauley, 2025-05-29:
When a child module contains a layer block, and multiple instances of that module get deduped, the `layers-PARENT_MODULE.sv` file contains one `include` directive for every instance of the child module. This doesn't cause any compilation issues, due to the `ifdef` guards in these files, but it would be nice for the emitted Verilog to be cleaner.

I was able to reproduce this on both `firtool-1.118.0` and `firtool-1.119.0`, but not on `firtool-1.117.0`. The format of the layer files changed significantly from 1.117.0 to 1.118.0, due to #8474 and #8458, so I'm not sure which one might have caused this regression.

Chisel version: `7.0.0-RC1+26-456303b4-SNAPSHOT`
firtool version: `firtool-1.119.0`

To reproduce:

```scala
//> using scala 2.13.16
//> using repository https://s01.oss.sonatype.org/content/repositories/snapshots
//> using dep org.chipsalliance::chisel:7.0.0-RC1+26-456303b4-SNAPSHOT
//> using plugin org.chipsalliance:::chisel-plugin:7.0.0-RC1+26-456303b4-SNAPSHOT
//> using options -unchecked -deprecation -language:reflectiveCalls -feature -Xcheckinit -Xfatal-warnings -Ywarn-dead-code -Ywarn-unused -Ymacro-annotations

import chisel3._

class Bar extends Module {
  printf("foo\n")
}

class Foo extends Module {
  val a = Module(new Bar)
  val b = Module(new Bar)
}

object Main extends App {
  import _root_.circt.stage.ChiselStage
  println(
    ChiselStage.emitSystemVerilog(
      new Foo,
      firtoolOpts = Array("-disable-all-randomization", "-strip-debug-info")
    )
  )
}
```

This results in:

```verilog
// Generated by CIRCT firtool-1.119.0
module Bar(
  input clock,
        reset
);

endmodule

module Foo(
  input clock,
        reset
);

  Bar a (
    .clock (clock),
    .reset (reset)
  );
  Bar b (
    .clock (clock),
    .reset (reset)
  );
endmodule

// ----- 8< ----- FILE "verification/Bar_Verification.sv" ----- 8< -----

// Generated by CIRCT firtool-1.119.0

// Users can define 'PRINTF_COND' to add an extra gate to prints.
`ifndef PRINTF_COND_
  `ifdef PRINTF_COND
    `define PRINTF_COND_ (`PRINTF_COND)
  `else  // PRINTF_COND
    `define PRINTF_COND_ 1
  `endif // PRINTF_COND
`endif // not def PRINTF_COND_
module Bar_Verification();
  `ifndef SYNTHESIS
    always @(posedge Bar.clock) begin
      if ((`PRINTF_COND_) & ~Bar.reset)
        $fwrite(32'h80000002, "foo\n");
    end // always @(posedge)
  `endif // not def SYNTHESIS
endmodule

// ----- 8< ----- FILE "verification/layers-Bar-Verification.sv" ----- 8< -----

// Generated by CIRCT firtool-1.119.0
`ifndef layers_Bar_Verification
  `define layers_Bar_Verification
  bind Bar Bar_Verification verification ();
`endif // not def layers_Bar_Verification

// ----- 8< ----- FILE "verification/layers-Foo-Verification.sv" ----- 8< -----

// Generated by CIRCT firtool-1.119.0
`ifndef layers_Foo_Verification
  `define layers_Foo_Verification
  `include "layers-Bar-Verification.sv"
  `include "layers-Bar-Verification.sv"
`endif // not def layers_Foo_Verification

// ----- 8< ----- FILE "verification/assert/layers-Foo-Verification-Assert.sv" ----- 8< -----

// Generated by CIRCT firtool-1.119.0
`ifndef layers_Foo_Verification_Assert
  `define layers_Foo_Verification_Assert
  `include "layers-Foo-Verification.sv"
`endif // not def layers_Foo_Verification_Assert

// ----- 8< ----- FILE "verification/assume/layers-Foo-Verification-Assume.sv" ----- 8< -----

// Generated by CIRCT firtool-1.119.0
`ifndef layers_Foo_Verification_Assume
  `define layers_Foo_Verification_Assume
  `include "layers-Foo-Verification.sv"
`endif // not def layers_Foo_Verification_Assume

// ----- 8< ----- FILE "verification/cover/layers-Foo-Verification-Cover.sv" ----- 8< -----

// Generated by CIRCT firtool-1.119.0
`ifndef layers_Foo_Verification_Cover
  `define layers_Foo_Verification_Cover
  `include "layers-Foo-Verification.sv"
`endif // not def layers_Foo_Verification_Cover
```

As you can see above, `verification/layers-Foo-Verification.sv` contains two `include` directives for `layers-Bar-Verification.sv`, but we would expect only one. The number of instances of the `Bar` module inside of `Foo` controls the number of `include` directives.

rwy7, 2025-05-30:
Thank you for the report! I'll take a look at this soon 👍

closed 2025-05-30 by rwy7

tymcauley, 2025-05-30:
Awesome, thanks for the quick fix!

PR #8521 (2025-05-30): [FIRRTL] LowerLayers: do not emit duplicate bindfile includes
When emitting bindfiles for a module, that bindfile will include the bindfiles of any module instantiated directly under it.

If a module is instantiated multiple times, we'll emit duplicate include statements. This PR modifies the bindfile creation code to avoid emitting duplicate include statements.

Fixes: #8519