llvm/circt #8525: Support for Adding Parameter Annotations to Generated Modules (e.g., as Comments)

farukyld, 2025-06-02:
Sory if I seem to come here without reading documentation, or without enough inspect of codebase. 

### Context
I'm using PyCDE to generate parameterized modules and test them using a handwritten Verilator testbench. To simplify the testbench logic, I instantiate different versions of the module under test using a single testbench. For that to work, all the modules must have the same top-level name.

To achieve this, I removed the parameters from the generated module name by setting an underscore ("_") as the name in the module builder. This works for Verilator, but it creates another problem…
### Problem
By removing the parameters from the module name, I lose visibility into what configuration a given module was generated from. This makes it difficult to track which parameters were used for which module, especially when dealing with multiple variations. As a result, debugging and traceability suffer.
### Question/Feature Request
Is there a way to annotate the generated Verilog/SystemVerilog module with a comment that includes the parameter values? Ideally, I’d like to see something like this:
```verilog
// Parameters: WIDTH=32, DEPTH=64
module my_module (...);
```
If this is already supported in PyCDE, could you provide an example of how to do it? If not, would it be feasible to add such a feature?

teqdruid, 2025-06-10:
Sorry it took me so long to post here.

Turns out there is a property in `hw.module` to emit a comment above. So this is a simple feature addition. I was thinking that the docstring should be put above the module. WDYT?

teqdruid, 2025-06-10:
Does it make sense to also add the module parameters? How about the metadata which you can optionally add?