llvm/circt #8532: SCF execute regions are improperly inlined into parent modules

fabianschuiki, 2025-06-03:
I have just come across the following `scf.execute_region` operation that is inlined into the parent module in a somewhat unsavory way:
```mlir
hw.module @Foo() {
  %c0_i5 = hw.constant 0 : i5
  %c-118_i8 = hw.constant -118 : i8
  %0 = llhd.constant_time <0ns, 0d, 1e>
  %c0_i153 = hw.constant 0 : i153
  %a = llhd.sig %c0_i153 : i153
  scf.execute_region {
    cf.br ^bb1
  ^bb1:  // pred: ^bb0
    llhd.drv %a, %c0_i153 after %0 : !hw.inout<i153>
    %1 = llhd.sig.extract %a from %c-118_i8 : (!hw.inout<i153>) -> !hw.inout<i5>
    llhd.drv %1, %c0_i5 after %0 : !hw.inout<i5>
    scf.yield
  }
  hw.output
}
```
Running this through `circt-opt --canonicalize` yields the following:
```mlir
hw.module @Foo() {
  %c0_i5 = hw.constant 0 : i5
  %c-118_i8 = hw.constant -118 : i8
  %0 = llhd.constant_time <0ns, 0d, 1e>
  %c0_i153 = hw.constant 0 : i153
  %a = llhd.sig %c0_i153 : i153
  // ----- 8< -----
  llhd.drv %a, %c0_i153 after %0 : !hw.inout<i153>
  %1 = llhd.sig.extract %a from %c-118_i8 : (!hw.inout<i153>) -> !hw.inout<i5>
  llhd.drv %1, %c0_i5 after %0 : !hw.inout<i5>
  // ----- 8< -----
  hw.output
}
```
I'm not quite sure what to make of this. The `llhd.drv` in this case have store-like side effects, and the second drive overwrites part of the value written by the first drive. This is valid in an SSACFG region like `scf.execute_region`, but inlining this into a graph region is problematic. Within a module the drives co-exist and actually produce a drive conflict on the bits of `%a` that are targeted by both drives.

@maerhart Do you have any ideas on how we could deal with something like this?

maerhart, 2025-06-04:
I think it's probably best to just create a new op that we can use instead of `execute_region` that has the exact semantics we need. We could forbid inlining of `execute_region` into graph region in it's canonicalizer, but that op is there for a slightly different reason than what we are doing, so we might run into other problems in the future and upstream might change things about this op that could lead to issues in CIRCT that we might not catch immediately.
I think we wanted to create an alternative, simpler version of `llhd.process` anyway to capture always_comb procedures without the complicated `llhd.wait` boilerplate. That could be a good match for this use-case as well, right?

fabianschuiki, 2025-06-04:
That's a great idea. This does sound like it could just be that special combinational process! And then we can have a more careful canonicalizer that only inlines into the parent if it's valid to do so. That might also be easier to handle in a control flow unrolling pass.

commit 4b8b11a7 (2025-06-05): [LLHD] Use new combinational process instead of execute regions

Make deseq and process lowering use the new `llhd.combinational` op
instead of the existing `scf.execute_region`. The latter is too eager to
inline into the parent region, even if the region contains ops with side
effects such as probes and drives. These no longer have a clear order in
the parent module. The specialized LLHD op is less eager about inlining
and can deal with the details of the SSACFG-to-graph region transition
in more detail.

This commit is basically just a whole bunch of renaming and changing
comments. No change [... cut]
commit f0255872 (2025-06-09): [LLHD] Use new combinational process instead of execute regions

Make deseq and process lowering use the new `llhd.combinational` op
instead of the existing `scf.execute_region`. The latter is too eager to
inline into the parent region, even if the region contains ops with side
effects such as probes and drives. These no longer have a clear order in
the parent module. The specialized LLHD op is less eager about inlining
and can deal with the details of the SSACFG-to-graph region transition
in more detail.

This commit is basically just a whole bunch of renaming and changing
comments. No change [... cut]
commit 3e03f617 (2025-06-09): [LLHD] Use new combinational process instead of execute regions (#8537)

Make deseq and process lowering use the new `llhd.combinational` op
instead of the existing `scf.execute_region`. The latter is too eager to
inline into the parent region, even if the region contains ops with side
effects such as probes and drives. These no longer have a clear order in
the parent module. The specialized LLHD op is less eager about inlining
and can deal with the details of the SSACFG-to-graph region transition
in more detail.

This commit is basically just a whole bunch of renaming and changing
comments. N [... cut]
closed 2025-06-09 by fabianschuiki