llvm/circt #8541: [FIRRTL] Nightly Failures after Bumping CI Images

seldridge, 2025-06-05:
A nightly failure was observed here: https://github.com/llvm/circt/actions/runs/15466476578

The last nightly passed on 0b3feee03. One of the following three commits is the culprit:

* 9be24fbb4 [LLHD] Fix canonicalizer crashing on empty processes with results
* 3e17a7391 [CI] Bump docker images (#8500)
* f3b535231 [Support] Factor out unused op pruning from LLHD Mem2Reg; NFC (#8531)

Given the failure is in a `firtool` end-to-end test, this almost has to be the docker image bump.

fzi-hielscher, 2025-06-05:
In case it helps: I ran the failing test (gcc release build without assertions) on one of our Ubuntu 24 machines without using the docker image, and the failure also appears there. Tesitng on commit  a7572c2a2:

`valgrind ./bin/firtool --repl-seq-mem --repl-seq-mem-file=mems.conf --split-verilog -o=./tmp ../test/Dialect/FIRRTL/SFCTests/ExtractSeqMems/Simple2.fir`

produces

```
==286676== Memcheck, a memory error detector
==286676== Copyright (C) 2002-2022, and GNU GPL'd, by Julian Seward et al.
==286676== Using Valgrind-3.22.0 and LibVEX; rerun with -h for copyright info
==286676== Command: ./bin/firtool --repl-seq-mem --repl-seq-mem-file=mems.conf --split-verilog -o=./tmp ../test/Dialect/FIRRTL/SFCTests/ExtractSeqMems/Simple2.fir
==286676==
==286676== Invalid read of size 8
==286676==    at 0x4AA28C4: (anonymous namespace)::ExtractInstancesPass::createTraceFiles(circt::firrtl::ClassOp&) (in /local_disk/hielsche/circt/build/lib/libCIRCTFIRRTLTransforms.so)
==286676==    by 0x4AAC43B: (anonymous namespace)::ExtractInstancesPass::runOnOperation() (in /local_disk/hielsche/circt/build/lib/libCIRCTFIRRTLTransforms.so)
==286676==    by 0x5BD3DCD: mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) (in /local_disk/hielsche/circt/build/lib/libMLIRPass.so)
==286676==    by 0x5BD41D8: mlir::detail::OpToOpPassAdaptor::runPipeline(mlir::OpPassManager&, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) (in /local_disk/hielsche/circt/build/lib/libMLIRPass.so)
==286676==    by 0x5BD331F: mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool) (in /local_disk/hielsche/circt/build/lib/libMLIRPass.so)
==286676==    by 0x5BD3B4D: mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) (in /local_disk/hielsche/circt/build/lib/libMLIRPass.so)
==286676==    by 0x5BD41D8: mlir::detail::OpToOpPassAdaptor::runPipeline(mlir::OpPassManager&, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) (in /local_disk/hielsche/circt/build/lib/libMLIRPass.so)
==286676==    by 0x5BD506A: mlir::PassManager::run(mlir::Operation*) (in /local_disk/hielsche/circt/build/lib/libMLIRPass.so)
==286676==    by 0x12C53E: processBuffer(mlir::MLIRContext&, circt::firtool::FirtoolOptions&, mlir::TimingScope&, llvm::SourceMgr&, std::optional<std::unique_ptr<llvm::ToolOutputFile, std::default_delete<llvm::ToolOutputFile> > >&) (in /local_disk/hielsche/circt/build/bin/firtool)
==286676==    by 0x12DEF7: processInputSplit(mlir::MLIRContext&, circt::firtool::FirtoolOptions&, mlir::TimingScope&, std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer> >, std::optional<std::unique_ptr<llvm::ToolOutputFile, std::default_delete<llvm::ToolOutputFile> > >&) (in /local_disk/hielsche/circt/build/bin/firtool)
==286676==    by 0x12EF8E: main (in /local_disk/hielsche/circt/build/bin/firtool)
==286676==  Address 0x10 is not stack'd, malloc'd or (recently) free'd
==286676==
PLEASE submit a bug report to https://github.com/llvm/circt and include the crash backtrace.
Stack dump:
0.      Program arguments: ./bin/firtool --repl-seq-mem --repl-seq-mem-file=mems.conf --split-verilog -o=./tmp ../test/Dialect/FIRRTL/SFCTests/ExtractSeqMems/Simple2.fir
 #0 0x0000000006013ef2 llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) (/local_disk/hielsche/circt/build/bin/../lib/libLLVMSupport.so+0x238ef2)
 #1 0x000000000601137e SignalHandler(int, siginfo_t*, void*) Signals.cpp:0:0
 #2 0x0000000006345330 (/lib/x86_64-linux-gnu/libc.so.6+0x45330)
 #3 0x0000000004aa28c4 (anonymous namespace)::ExtractInstancesPass::createTraceFiles(circt::firrtl::ClassOp&) ExtractInstances.cpp:0:0
 #4 0x0000000004aac43c (anonymous namespace)::ExtractInstancesPass::runOnOperation() ExtractInstances.cpp:0:0
 #5 0x0000000005bd3dce mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) (/local_disk/hielsche/circt/build/bin/../lib/libMLIRPass.so+0x28dce)
 #6 0x0000000005bd41d9 mlir::detail::OpToOpPassAdaptor::runPipeline(mlir::OpPassManager&, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) (/local_disk/hielsche/circt/build/bin/../lib/libMLIRPass.so+0x291d9)
 #7 0x0000000005bd3320 mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool) (/local_disk/hielsche/circt/build/bin/../lib/libMLIRPass.so+0x28320)
 #8 0x0000000005bd3b4e mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) (/local_disk/hielsche/circt/build/bin/../lib/libMLIRPass.so+0x28b4e)
 #9 0x0000000005bd41d9 mlir::detail::OpToOpPassAdaptor::runPipeline(mlir::OpPassManager&, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) (/local_disk/hielsche/circt/build/bin/../lib/libMLIRPass.so+0x291d9)
#10 0x0000000005bd506b mlir::PassManager::run(mlir::Operation*) (/local_disk/hielsche/circt/build/bin/../lib/libMLIRPass.so+0x2a06b)
#11 0x000000000012c53f processBuffer(mlir::MLIRContext&, circt::firtool::FirtoolOptions&, mlir::TimingScope&, llvm::SourceMgr&, std::optional<std::unique_ptr<llvm::ToolOutputFile, std::default_delete<llvm::ToolOutputFile>>> [... cut ...]
==286676== For lists of detected and suppressed errors, rerun with: -s
==286676== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)
Segmentation fault (core dumped)
```

On first glance, it seems like  `metadataObj` in ExtractInstances is `null` for some reason:
https://github.com/llvm/circt/blob/f9f0c2be95017a835e4bb729cb9201d64469f6b0/lib/Dialect/FIRRTL/Transforms/ExtractInstances.cpp#L1156-L1159

For reference:
```
Arch: x86_64
OS: Ubuntu 24.04.2 LTS
GCC: Ubuntu 13.3.0-6ubuntu2~24.04
GLIBC: Ubuntu GLIBC 2.39-0ubuntu8.4
```

fzi-hielscher, 2025-06-06:
I don't have the time to really dig into it right now, so take this with a big grain of salt: It might be related to https://github.com/llvm/llvm-project/issues/86647. Basically, at some point the compiler decides that the `InstanceOpInterface` result type of `(*node->usesBegin())->getInstance()` can never be cast to `ObjectOp` and thus `metadataObj` will always be `null`. The fix seems to be pleasingly simple:
```diff
diff --git a/lib/Dialect/FIRRTL/Transforms/ExtractInstances.cpp b/lib/Dialect/FIRRTL/Transforms/ExtractInstances.cpp
index 8e6c16ef4..b040984dd 100644
--- a/lib/Dialect/FIRRTL/Transforms/ExtractInstances.cpp
+++ b/lib/Dialect/FIRRTL/Transforms/ExtractInstances.cpp
@@ -1153,8 +1153,7 @@ void ExtractInstancesPass::createTraceFiles(ClassOp &sifiveMetadataClass) {
     addPortsToClass(classFields, sifiveMetadataClass);
     auto *node = instanceGraph->lookup(sifiveMetadataClass);
     assert(node && node->hasOneUse());
-    ObjectOp metadataObj =
-        dyn_cast_or_null<ObjectOp>((*node->usesBegin())->getInstance());
+    ObjectOp metadataObj = (*node->usesBegin())->getInstance<ObjectOp>();
     assert(metadataObj &&
            "expected the class to be instantiated by an object op");
     builderOM.setInsertionPoint(metadataObj);
```

seldridge, 2025-06-06:
Wow, thanks! I'm giving that a try here: https://github.com/llvm/circt/actions/runs/15493581724

seldridge, 2025-06-06:
Fixed via https://github.com/llvm/circt/commit/de9b2b9002bee4659406727908fc3dc8e3b188f8.

closed 2025-06-06 by seldridge