llvm/circt #8653: [FIRRTL] Add InferTypeOpInterface to firrtl.int.ltl.* Operations

sequencer, 2025-07-06:
Description:
---
The operations within the `firrtl.int.ltl.*` namespace, such as `firrtl.int.ltl.and`, `firrtl.int.ltl.or`, and `firrtl.int.ltl.eventually`, represent LTL intrinsics which always produce a boolean (`i1`) result.

These operations currently lack the `InferTypeOpInterface`, requiring users of the C API to manually specify the result type during creation.

Proposal:
---
Implement the `InferTypeOpInterface` for all `firrtl.int.ltl.*` operations to automatically infer their `i1` result type.

The following `firrtl.int.ltl.*` operations should be updated to implement the `InferTypeOpInterface`:
```
    firrtl.int.ltl.and
    firrtl.int.ltl.or
    firrtl.int.ltl.not
    firrtl.int.ltl.implication
    firrtl.int.ltl.eventually
    firrtl.int.ltl.clock
    firrtl.int.ltl.concat
    firrtl.int.ltl.delay
    firrtl.int.ltl.goto_repeat
    firrtl.int.ltl.intersect
    firrtl.int.ltl.non_consecutive_repeat
    firrtl.int.ltl.repeat
    firrtl.int.ltl.until
```

cc @fabianschuiki

seldridge, 2025-07-07:
Makes sense to me.

fabianschuiki, 2025-07-08:
Yeah that sounds like a good idea! I'm surprised that this doesn't already work. The `FIRRTLExpressions.td` file, where the LTL ops are defined, contains a `include "mlir/Interfaces/InferTypeOpInterface.td"`, which together with the fixed result type in `let results = (outs UInt1Type:$result);` should be enough to infer the result type. But maybe not for the C API? If I remember correctly, the C++ API should not require you to specify a result type -- it infers it automatically.

fabianschuiki, 2025-07-08:
Ah I see the problem. We have to use `NonConstUInt1Type` instead of `UInt1Type`:
```tablegen
let results = (outs NonConstUInt1Type:$result);
```
That then creates builders that don't have an explicit result type operand:
```c++
// from FIRRTL.h.inc
// class LTLAndIntrinsicOp
static void build(OpBuilder &odsBuilder, OperationState &odsState, Value lhs, Value rhs);
```

commit ceec7290 (2025-07-08): [FIRRTL] Use NonConstUInt1Type for LTL intrinsics

Change the result type of LTL intrinsic ops from `UInt1Type` to
`NonConstUInt1Type`. The latter defines a default builder, which then
allows the `build` method of ops like `LTLAndIntrinsicOp` to drop the
result type argument and automatically populate it with `uint<1>`. This
should then also simplify the C API.

This only adds an additional build method and shouldn't change any
existing use case.

Thanks @sequencer for pointing this out!

Fixes #8653.