llvm/circt #8657: Imported verilog flips output

jpienaar, 2025-07-07:
```verilog
module conc(
  input wire [63:0] arg0,
  output wire [63:0] result0
);
  wire [31:0] arg0_unflattened[2];
  assign arg0_unflattened[0] = arg0[31:0];
  assign arg0_unflattened[1] = arg0[63:32];

  assign result0 = {arg0_unflattened[1], arg0_unflattened[0]};
endmodule
```

Post

```
./bin/circt-verilog | ./bin/circt-opt --export-verilog
```

we get

```verilog
module conc(	// <stdin>:2:3
  input  [63:0] arg0,	// <stdin>:2:22
  output [63:0] result0	// <stdin>:2:39
);
module conc(

  assign result0 = {arg0[31:0], arg0[63:32]};	// <stdin>:3:10, :4:10, :5:10, :6:5
endmodule
```

(doing it inlined rather than assigning to unflattened first results in result0 = argo)

uenoku, 2025-07-07:
It seems like related to unpacked array handling in ImportVerilog. It works fine for packed array. 

```verilog
$ cat bar.sv
module conc(
  input wire [63:0] arg0,
  output wire [63:0] result0
);
  wire [1:0][31:0] arg0_unflattened;
  assign arg0_unflattened[0] = arg0[31:0];
  assign arg0_unflattened[1] = arg0[63:32];

  assign result0 = {arg0_unflattened[1], arg0_unflattened[0]};
endmodule

$ circp-verilog bar.sv | circt-opt -export-verilog
// Generated by CIRCT firtool-1.124.0-12-g4bb54cdf9-dirty
module conc(    // <stdin>:2:3
  input  [63:0] arg0,   // <stdin>:2:22
  output [63:0] result0 // <stdin>:2:39
);

  wire [31:0] _arg0_63to32;     // <stdin>:5:10
  wire [31:0] _arg0_31to0;      // <stdin>:4:10
  assign _arg0_31to0 = arg0[31:0];      // <stdin>:4:10
  assign _arg0_63to32 = arg0[63:32];    // <stdin>:5:10
  assign result0 = /*cast(bit[63:0])*/{{_arg0_63to32}, {_arg0_31to0}};  // <stdin>:3:10, :4:10, :5:10, :6:10, :7:5
endmodule

```

jpienaar, 2025-07-08:
Which is weird ... if I look at ImportVerilog, nothing jumps out. The two different forms (packed and unpacked probably the closest).

The two variants just post import:

```mlir
module {
  moore.module @conc(in %arg0 : !moore.l64, out result0 : !moore.l64) {
    %arg0_0 = moore.net name "arg0" wire : <l64>
    %result0 = moore.net wire : <l64>
    %arg0_unflattened = moore.net wire : <array<2 x l32>>
    %0 = moore.extract_ref %arg0_unflattened from 0 : <array<2 x l32>> -> <l32>
    %1 = moore.read %arg0_0 : <l64>
    %2 = moore.extract %1 from 0 : l64 -> l32
    moore.assign %0, %2 : l32
    %3 = moore.extract_ref %arg0_unflattened from 1 : <array<2 x l32>> -> <l32>
    %4 = moore.read %arg0_0 : <l64>
    %5 = moore.extract %4 from 32 : l64 -> l32
    moore.assign %3, %5 : l32
    %6 = moore.read %arg0_unflattened : <array<2 x l32>>
    %7 = moore.extract %6 from 1 : array<2 x l32> -> l32
    %8 = moore.read %arg0_unflattened : <array<2 x l32>>
    %9 = moore.extract %8 from 0 : array<2 x l32> -> l32
    %10 = moore.concat %7, %9 : (!moore.l32, !moore.l32) -> l64
    moore.assign %result0, %10 : l64
    moore.assign %arg0_0, %arg0 : l64
    %11 = moore.read %result0 : <l64>
    moore.output %11 : !moore.l64
  }
}
```

vs

```mlir
module {
  moore.module @conc(in %arg0 : !moore.l64, out result0 : !moore.l64) {
    %arg0_0 = moore.net name "arg0" wire : <l64>
    %result0 = moore.net wire : <l64>
    %arg0_unflattened = moore.net wire : <uarray<2 x l32>>
    %0 = moore.extract_ref %arg0_unflattened from 0 : <uarray<2 x l32>> -> <l32>
    %1 = moore.read %arg0_0 : <l64>
    %2 = moore.extract %1 from 0 : l64 -> l32
    moore.assign %0, %2 : l32
    %3 = moore.extract_ref %arg0_unflattened from 1 : <uarray<2 x l32>> -> <l32>
    %4 = moore.read %arg0_0 : <l64>
    %5 = moore.extract %4 from 32 : l64 -> l32
    moore.assign %3, %5 : l32
    %6 = moore.read %arg0_unflattened : <uarray<2 x l32>>
    %7 = moore.extract %6 from 0 : uarray<2 x l32> -> l32
    %8 = moore.read %arg0_unflattened : <uarray<2 x l32>>
    %9 = moore.extract %8 from 1 : uarray<2 x l32> -> l32
    %10 = moore.concat %7, %9 : (!moore.l32, !moore.l32) -> l64
    moore.assign %result0, %10 : l64
    moore.assign %arg0_0, %arg0 : l64
    %11 = moore.read %result0 : <l64>
    moore.output %11 : !moore.l64
  }
}
```

So the extracts from unflattened differ, while concat remains the same. For the unpacked one this is the visitor state

```
convertRvalueExpression: RangeSelect
RangeSelectExpression
convertRvalueExpression: NamedValue
        RangeSelectExpression: Simple
                lhs: 31, rhs: 0
convertRvalueExpression: RangeSelect
RangeSelectExpression
convertRvalueExpression: NamedValue
        RangeSelectExpression: Simple
                lhs: 63, rhs: 32
convertRvalueExpression: Concatenation
ConcatenationExpression
convertRvalueExpression: ElementSelect
convertRvalueExpression: NamedValue
        range = [0:1]
        ElementSelectExpression: 1, translated index: 0
convertRvalueExpression: ElementSelect
convertRvalueExpression: NamedValue
        range = [0:1]
        ElementSelectExpression: 0, translated index: 1
```

while for packed

```
convertRvalueExpression: RangeSelect
RangeSelectExpression
convertRvalueExpression: NamedValue
        RangeSelectExpression: Simple
                lhs: 31, rhs: 0
convertRvalueExpression: RangeSelect
RangeSelectExpression
convertRvalueExpression: NamedValue
        RangeSelectExpression: Simple
                lhs: 63, rhs: 32
convertRvalueExpression: Concatenation
ConcatenationExpression
convertRvalueExpression: ElementSelect
convertRvalueExpression: NamedValue
        range = [1:0]
        ElementSelectExpression: 1, translated index: 1
convertRvalueExpression: ElementSelect
convertRvalueExpression: NamedValue
        range = [1:0]
        ElementSelectExpression: 0, translated index: 0
```

So I get same flipping if I do `wire [0:1][31:0] arg0_unflattened;`. So can get different results for two packed ones.

jpienaar, 2025-07-08:
I do notice one thing: RvalueExprVisitor and LvalueExprVisitor differ when visiting ElementSelectExpression, the former does a range based translation, while the latter does not. And if I change the Rvalue one to match the Lvalue one, then I do get the expected result.

jpienaar, 2025-07-08:
Same too if I make Lvalue one match Rvalue one and that seems correct change direction.

commit b033712e (2025-07-08): [importverilog] Lvalue RangelSelect and ElementSelect to use range. (#8663)

This makes it uniform with how the RvalueExprVisitor handles it.

Resolves: #8657
closed 2025-07-08 by jpienaar