llvm/circt #8672: [FIRRTL] BlackBoxTargetDirAnno not respected for all black boxes

tymcauley, 2025-07-09:
This came up when looking into using `firrtl.transforms.BlackBoxTargetDirAnno` after #8669 was merged. Unfortunately I couldn't figure out a minimal example to recreate this issue. I'm attaching a Rocket Chip design generated from a Chipyard fork which is using Chisel 7.0.0-RC2 and `firtool-1.124.0`. When I add the following annotation to the design:

```json
  {
    "class":"firrtl.transforms.BlackBoxTargetDirAnno",
    "targetDir":"black-boxes"
  }
```

I would expect all of the black box files to be placed in the `black-boxes` subdirectory inside the `firtool` output directory. However, when I run this command:

```shell
firtool --annotation-file=rocket.annos.json --split-verilog -o out-dir rocket.fir
```

I see the following:

```shell
$ cat out-dir/firrtl_black_box_resource_files.f
black-boxes/ClockSourceAtFreqMHz.v
black-boxes/EICG_wrapper.v
black-boxes/GenericDigitalInIOCell.v
black-boxes/GenericDigitalOutIOCell.v
black-boxes/SimDRAM.cc
black-boxes/SimDRAM.v
black-boxes/SimJTAG.cc
black-boxes/SimJTAG.v
black-boxes/SimTSI.cc
black-boxes/SimTSI.v
black-boxes/SimUART.cc
black-boxes/SimUART.v
black-boxes/mm.cc
black-boxes/mm_dramsim2.cc
black-boxes/remote_bitbang.cc
black-boxes/testchip_htif.cc
black-boxes/testchip_tsi.cc
black-boxes/uart.cc
plusarg_reader.v
```

Notably, `plusarg_reader.v` is placed directly in `out-dir`, rather than in the `black-boxes` subdirectory. Any ideas why this might be happening? You can recreate this by downloading the attached archive, and then running:

```shell
$ tar xf rocket-chip-black-box-path-error.tar.gz
$ cd rocket-chip-black-box-path-error/
$ ./run.sh
```

[rocket-chip-black-box-path-error.tar.gz](https://github.com/user-attachments/files/21134761/rocket-chip-black-box-path-error.tar.gz)

seldridge, 2025-07-09:
There may be an issue/interaction involving blackbox "resources". In Chisel resource blackboxes are converted to inline blackboxes before generating FIRRTL. There may be a problem with how the annotation is setup (e.g., an absolute path).

I'm going to check how each of inline, path, and resource blackboxes behave. A shot in the dark you could try is to change `plusarg_reader` to an inline blackbox and see if it fixes it here: https://github.com/chipsalliance/rocket-chip/blob/f517abbf41abb65cea37421d3559f9739efd00a9/src/main/scala/util/PlusArg.scala#L45

seldridge, 2025-07-09:
Okay, I see roughly what is going on. High level, `BlackBoxTargetDirAnno` can be overridden based on how modules are used. This blackbox winds up under a layerblock (due to a bug in the naive layer sink pass that is fixed with `-advanced-layer-sink`). However, the problem can persist if you manually instantiate the black box under a layer block.

More generally, this is having to make a call about where to put the blackboxes (generally modules) when they have conflicting information about where they should go. For this example, this is:

1. `./black-boxes/`
2. `./verification/assert/`

The rule here is that if this happens, the module will go in the lowest-common-ancestor directory, i.e., `./`.

I only have non-solutions as the directory constraints can't easily be resolved for all users (sometimes you want one directory, sometimes you want another):

1. Don't sink the blackbox and use `-advanced-layer-sink` (which, thanks to @rwy7 strictly understands that blackboxes can have unknown side effects), or use `-disable-layer-sink`. `-advanced-layer-sink` will soon become the default as opposed to the current "naive" layer sink that is on by default.
2. Figure out a way to restructure the directories so that they make sense.
3. Add a parameter to the blackbox generator that lets you give it a unique name so that you can use different names in different contexts.
4. Switch to the plus arg intrinsic and don't use a black box.
5. Understand that this is how the algorithm works and not all blackboxes can be sequestered like this. (Or: is it really important that the blackboxes are sequestered in a directory given that they are no different from any other Verilog that Chisel generates?)

### Example

Consider the following circuit. This is telling the compiler to create a file called `inline.sv` and put it in both `verification/assert/` and `black-boxes/`. These are incompatible so it puts it in the parent:

``` firrtl
FIRRTL version 5.1.0
circuit Foo :%[[
	{
		"class":"sifive.enterprise.firrtl.MarkDUTAnnotation",
		"target":"~|Foo"
	},
  {
    "class":"firrtl.transforms.BlackBoxInlineAnno",
    "target":"~|Inline",
    "name":"inline.sv",
    "text":"module Inline();\nendmodule\n"
  },
  {
    "class":"firrtl.transforms.BlackBoxInlineAnno",
    "target":"~|Inline2",
    "name":"inline.sv",
    "text":"module Inline();\nendmodule\n"
  },
  {
    "class":"firrtl.transforms.BlackBoxTargetDirAnno",
    "targetDir":"black-boxes"
  }
]]
  layer Verification, bind, "verification" :
    layer Assert, bind, "verification/assert" :

  extmodule Inline :
    defname = Inline
    parameter A = 0

  extmodule Inline2 :
    defname = Inline
    parameter A = 2

  public module Foo :
    input clock : Clock
    input reset : UInt<1>

    inst foo of Inline
    layerblock Verification:
      layerblock Assert:
        inst bar of Inline2
```

Full output with pass debugging information:

```
# firtool circt-8672.fir -debug-only=firrtl-blackbox-reader
===- Running BlackBoxReader -------------------------------------------------===
Black box target directory: black-boxes
Visiting extmodules:
  - name: "Inline"
    annotations:
      - name: inline.sv
        outputFile: black-boxes/inline.sv
        exclude: false
  - name: "Inline2"
    annotations:
      - name: inline.sv
        outputFile: verification/assert/inline.sv
        exclude: false
emittedFiles:
  verilogName: "inline.sv"
  annotationInfo:
    - name: inline.sv
      outputFile: inline.sv
      exclude: false
===--------------------------------------------------------------------------===
// Generated by CIRCT firtool-1.124.0-40-g713a91ddf
module Foo(
  input clock,
        reset
);

  Inline #(
    .A(0)
  ) foo ();
endmodule

// ----- 8< ----- FILE "verification/assert/Foo_Verification_Assert.sv" ----- 8< -----

// Generated by CIRCT firtool-1.124.0-40-g713a91ddf
module Foo_Verification_Assert();
  Inline #(
    .A(2)
  ) bar ();
endmodule

// ----- 8< ----- FILE "verification/Foo_Verification.sv" ----- 8< -----

// Generated by CIRCT firtool-1.124.0-40-g713a91ddf
module Foo_Verification();
endmodule

// ----- 8< ----- FILE "verification/layers-Foo-Verification.sv" ----- 8< -----

// Generated by CIRCT firtool-1.124.0-40-g713a91ddf
`ifndef layers_Foo_Verification
  `define layers_Foo_Verification
  bind Foo Foo_Verification verification ();
`endif // not def layers_Foo_Verification

// ----- 8< ----- FILE "verification/assert/layers-Foo-Verification-Assert.sv" ----- 8< -----

// Generated by CIRCT firtool-1.124.0-40-g713a91ddf
`ifndef layers_Foo_Verification_Assert
  `define layers_Foo_Verification_Assert
  `include "layers-Foo-Verification.sv"
  bind Foo Foo_Verification_Assert verification_assert ();
`endif // not def layers_Foo_Verification_Assert

// ----- 8< ----- FILE "inline.sv" ----- 8< -----

// Generated by CIRCT firtool-1.124.0-40-g713a91ddf
module Inline();
endmodule%    
```