llvm/circt #8677: Importing Verilog snippets and $finish equivalent in PyCDE

AnZhong24, 2025-07-09:
Dear all, 
 
I am exploring **PyCDE** and was wondering how to best integrate existing Verilog code into a PyCDE project.

Specifically, I would like to know if it's possible to **import or inline a Verilog snippet** like the ifdef block below. Is there a recommended API for inserting existing Verilog code?
 
`ifndef SYNTHESIS
  always_ff @(posedge clk)
    if (self.executed && {0}) $finish;
endif`

Additionally, could you let me know if PyCDE provides a direct API equivalent for the Verilog system task **$finish** for use in simulations?

Thank you for your time.

teqdruid, 2025-07-11:
> Specifically, I would like to know if it's possible to **import or inline a Verilog snippet** like the ifdef block below. Is there a recommended API for inserting existing Verilog code?

In the PyCDE API, there's not. But you could reach underneath the hood and use the CIRCT API (which PyCDE uses internally) to create one or more [`VerbatimOp`s](https://circt.llvm.org/docs/Dialects/SV/#svverbatim-circtsvverbatimop).

```
class Mux(Module):
  clk = Clock()
  data = Input(dim(8, 14))
  sel = Input(Bits(4))

  out = Output(Bits(8))

  @generator
  def build(ports):
    from pycde.dialects import sv
    sv.VerbatimOp("""
`ifndef SYNTHESIS
  always_ff @(posedge clk)
    if (self.executed && {{0}}) $finish;
`endif
    """,
       substitutions=[ports.sel.value])
```

The verbatim op is just to inline text (with substitutions). There is not a way to import SV text.

> Additionally, could you let me know if PyCDE provides a direct API equivalent for the Verilog system task **$finish** for use in simulations?

Not currently. I've been meaning to add some simulation constructs like `$display` though. Just hasn't bubbled to the top of my priority list.

AnZhong24, 2025-07-11:
This is exactly what I needed to know. Thanks for clarifying!