llvm/circt #8692: [ExportVerilog] Support `math.ipowi` operation

fabianschuiki, 2025-07-11:
The following Verilog currently fails to round-trip through `circt-verilog | circt-opt --lower-seq-to-sv --export-verilog`:
```systemverilog
module binary_ops_00000105(a, y);
  input [3:0] a;
  output [4:0] y;
  assign y = 4'd2 ** a;
endmodule
```
Error:
```
syn_circt/binary_ops_00000105.mlir:6:10: error: dialect "math" not supported for direct Verilog emission
    %1 = math.ipowi %c2_i6, %0 : i6
         ^
syn_circt/binary_ops_00000105.mlir:6:10: note: see current operation: %3 = "math.ipowi"(%0, %2) : (i6, i6) -> i6
syn_circt/binary_ops_00000105.mlir:6:10: note: ExportVerilog cannot emit this operation; it needs to be lowered before running ExportVerilog
```
I don't think we have ever run into a need to have `**` emitted back to Verilog before. There is no op in the Comb dialect to represent `**`, which is why ImportVerilog uses `math.ipowi`. It might be worthwhile adding direct support for `math.ipowi` to ExportVerilog for the few cases where this comes up.

uenoku, 2025-07-12:
IMO ExportVerilog shouldn't have dependency to math dialect. Because `**` is operator there are several places we need to modify. I'm surprised SV has `**` operator but from that view point I think it's ok for comb to have `ipowi` operation (though I have no idea how to synthesize this operation or practical use case). Alternative solution is to legalize math.ipowi to `$pow` system verilog builtin function in PrepareForEmission.

fabianschuiki, 2025-07-15:
That's a good thought. Something like `sv.pow` sounds pretty reasonable -- that way we can deal with it outside the exporter, and have a dedicated preparation/lowering pass deal with figuring out the exact semantics of it.

Synthesis tools will probably treat `**` a bit like `/`: I'm pretty sure they do _some_ lowering, but in practice you'd probably always want a custom-built iterative architecture.