llvm/circt #8693: llhd mem2reg failure

jpienaar, 2025-07-11:
For input (derived from SV import minized and then MLIR minized)

```mlir
module {
  hw.module @a() {
    %false = hw.constant false
    %b = llhd.sig %false : i1
    llhd.combinational {
      cf.br ^bb2
    ^bb1:  // no predecessors
      %0 = llhd.prb %b : !hw.inout<i1>
      %1 = llhd.constant_time <0ns, 0d, 1e>
      %g = llhd.sig %false : i1
      llhd.drv %g, %0 after %1 : !hw.inout<i1>
      cf.br ^bb2
    ^bb2:  // 2 preds: ^bb0, ^bb1
      llhd.yield
    }
    hw.output
  }
}
```

results post llhd-mem2reg in

```
error: operand #0 does not dominate this use
note: see current operation: "llhd.drv"(%6, %7, %9, %8) : (!hw.inout<i1>, i1, !llhd.time, i1) -> ()
/tmp/t2.mlir:10:12: note: operand defined here (op in the same region)
      %g = llhd.sig %false : i1
           ^
```

print IR after:

```mlir
"builtin.module"() ({
  "hw.module"() <{module_type = !hw.modty<>, parameters = [], sym_name = "a"}> ({
    %0 = "hw.constant"() <{value = false}> : () -> i1
    %1 = "llhd.sig"(%0) <{name = "b"}> : (i1) -> !hw.inout<i1>
    %2 = "llhd.prb"(%1) : (!hw.inout<i1>) -> i1
    "llhd.combinational"() ({
      %3 = "hw.constant"() <{value = false}> : () -> i1
      %4 = "hw.constant"() <{value = false}> : () -> i1
      "cf.br"(%3, %4)[^bb2] : (i1, i1) -> ()
    ^bb1:  // no predecessors
      %5 = "hw.constant"() <{value = true}> : () -> i1
      %6 = "llhd.sig"(%0) <{name = "g"}> : (i1) -> !hw.inout<i1>
      "cf.br"(%2, %5)[^bb2] : (i1, i1) -> ()
    ^bb2(%7: i1, %8: i1):  // 2 preds: ^bb0, ^bb1
      %9 = "llhd.constant_time"() <{value = #llhd.time<0ns, 0d, 1e>}> : () -> !llhd.time
      "llhd.drv"(%6, %7, %9, %8) : (!hw.inout<i1>, i1, !llhd.time, i1) -> ()
      // ^
      "llhd.yield"() : () -> ()
    }) : () -> ()
    "hw.output"() : () -> ()
  }) : () -> ()
}) : () -> ()
```

fabianschuiki, 2025-07-11:
Very nice find! This is a pretty interesting failure 🤔. It looks like the problem is that Mem2Reg tries to insert a final drive to *g* at the end of the process, but *g* does not dominate that point. I'm wondering whether we could only propagate signal definitions to blocks and ops that are dominated by the signal. In this case, the definition for *g* would not propagate into `^bb2`, which causes the drive to be inserted at the end of `^bb1` instead.

seldridge, 2026-03-06:
@thomasnormal, @thomasahle: Due to current CIRCT project policy, we are unable to accept patches from the `thomasnormal/circt` fork.

For others: please do not upstream any patches from the `thomasnormal/circt` fork. Additionally, recognize that any comments made by the @thomasnormal account appear to be from one or more AI agents.

For more information, please see this Discourse post: https://discourse.llvm.org/t/circt-project-policy-on-contributions-from-thomasnormal-circt-fork/90075