llvm/circt #8699: Bug Report: Segmentation Fault in PyCDE When Instantiating a Struct with Constant Values

AnZhong24, 2025-07-12:
Dear all , the segmentation fault remains to occur during the system.compile() phase when a PyCDE module instantiates a Struct with constant literal values.  
 
```
from pycde import  Input, Output, generator, System, Module
from pycde.types import  Bits,UInt
from pycde.signals import Struct 
 

class ExStruct(Struct):
  a: Bits(4)
  b: UInt(32)
 
 
class TestStruct(Module):
  inp1 = Input(ExStruct)  
  @generator
  def build(self):
   
    # s = ExStruct(a=Bits(4)(0), b=self.inp1.b)
    s =  ExStruct(a=Bits(4)(0), b=UInt(32)(0)) 
     

system = System([TestStruct], name="Top", output_directory="sv")
system.compile()
```
 

 ```

Stack dump without symbol names (ensure you have llvm-symbolizer in your PATH or set the environment var `LLVM_SYMBOLIZER_PATH` to point to it):
0  libPyCDE_CIRCTPythonCAPI.so           0x00007f98db4a32d8 llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) + 40
1  libPyCDE_CIRCTPythonCAPI.so           0x00007f98db49fbf5
2  libc.so.6                             0x00007f98defd5520
3  libPyCDE_CIRCTPythonCAPI.so           0x00007f98db58b590 mlir::Block::getParentOp() + 0
4  libPyCDE_CIRCTPythonCAPI.so           0x00007f98dbe59cf4 circt::hw::HWDialect::materializeConstant(mlir::OpBuilder&, mlir::Attribute, mlir::Type, mlir::Location) + 116
5  libPyCDE_CIRCTPythonCAPI.so           0x00007f98db594d42 mlir::OpBuilder::tryFold(mlir::Operation*, llvm::SmallVectorImpl<mlir::Value>&, llvm::SmallVectorImpl<mlir::Operation*>*) + 1090
6  libPyCDE_CIRCTPythonCAPI.so           0x00007f98dc424810
7  libPyCDE_CIRCTPythonCAPI.so           0x00007f98dc424ed3 mlir::OperationConverter::convert(mlir::ConversionPatternRewriter&, mlir::Operation*) + 35
8  libPyCDE_CIRCTPythonCAPI.so           0x00007f98dc426904 mlir::OperationConverter::convertOperations(llvm::ArrayRef<mlir::Operation*>) + 308
9  libPyCDE_CIRCTPythonCAPI.so           0x00007f98dc428df3 mlir::applyPartialConversion(mlir::Operation*, mlir::ConversionTarget const&, mlir::FrozenRewritePatternSet const&, mlir::ConversionConfig) + 131
10 libPyCDE_CIRCTPythonCAPI.so           0x00007f98dc1586a1
11 libPyCDE_CIRCTPythonCAPI.so           0x00007f98db6a5f91 mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) + 1249
12 libPyCDE_CIRCTPythonCAPI.so           0x00007f98db6a62e9 mlir::detail::OpToOpPassAdaptor::runPipeline(mlir::OpPassManager&, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) + 233
13 libPyCDE_CIRCTPythonCAPI.so           0x00007f98db6a666e
14 libPyCDE_CIRCTPythonCAPI.so           0x00007f98db6a53b5 mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool) + 3909
15 libPyCDE_CIRCTPythonCAPI.so           0x00007f98db6a5d20 mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) + 624
16 libPyCDE_CIRCTPythonCAPI.so           0x00007f98db6a62e9 mlir::detail::OpToOpPassAdaptor::runPipeline(mlir::OpPassManager&, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) + 233
17 libPyCDE_CIRCTPythonCAPI.so           0x00007f98db6a7054 mlir::PassManager::run(mlir::Operation*) + 1300
18 libPyCDE_CIRCTPythonCAPI.so           0x00007f98db4405d9 mlirPassManagerRunOnOp + 9
19 _mlir.cpython-313-x86_64-linux-gnu.so 0x00007f98de75744c
20 _mlir.cpython-313-x86_64-linux-gnu.so 0x00007f98de75a016
21 python                                0x0000562d7c79989e PyObject_Vectorcall + 46
22 python                                0x0000562d7c687d79
23 python                                0x0000562d7c853091 PyEval_EvalCode + 161
24 python                                0x0000562d7c877ca0
25 python                                0x0000562d7c872bdc
26 python                                0x0000562d7c891a86
27 python                                0x0000562d7c890518
28 python                                0x0000562d7c8902ac
29 python                                0x0000562d7c889dce Py_RunMain + 814
30 python                                0x0000562d7c842fe7 Py_BytesMain + 55
31 libc.so.6                             0x00007f98defbcd90
32 libc.so.6                             0x00007f98defbce40 __libc_start_main + 128
33 python                                0x0000562d7c8423de
PLEASE submit a bug report to https://github.com/llvm/llvm-project/issues/ and include the crash backtrace.
Segmentation fault (core dumped)
```

teqdruid, 2025-07-12:
I don't get a segfault when I run that code. Are you running against the latest CIRCT main? I don't think the fix has made it into a proper release yet.

teqdruid, 2025-07-12:
I'm pushing another release now. Just looked and it's been two months since I did a release.