llvm/circt #8771: [SV] ETC crash with signed printf operand

dtzSiFive, 2025-07-23:
Input:
```firrtl
FIRRTL version 5.1.0
circuit SignedPrintETC :

  public module SignedPrintETC :
    input clk : Clock
    input test : SInt<64>

    printf(clk, UInt<1>(0h1), "%0d", test)
```

When using `--extract-test-code` this produces invalid IR:

```
signed-print-etc.fir:4:10: error: operand #0 does not dominate this use
  public module SignedPrintETC :
         ^
signed-print-etc.fir:4:10: note: see current operation: "hw.instance"(%2, %1, %0) <{argNames = ["port_0", "port_1", "port_2"], doNotPrint, inner_sym = #hw<innerSym@__ETC_SignedPrintETC_assert>, instanceName = "SignedPrintETC_assert", moduleName = @SignedPrintETC_assert, parameters = [], resultNames = []}> : (i64, i1, i1) -> ()
signed-print-etc.fir:8:5: note: operand defined here (op in a child region)
    printf(clk, UInt<1>(0h1), "%0d", test)
    ^
```

dtzSiFive, 2025-07-23:
MLIR after failure:
```mlir
// -----// IR Dump After SVExtractTestCode Failed (sv-extract-test-code) //----- //
"builtin.module"() ({
  "sv.macro.decl"() <{sym_name = "SYNTHESIS"}> : () -> ()
  "sv.macro.decl"() <{sym_name = "PRINTF_COND"}> : () -> ()
  "sv.macro.decl"() <{sym_name = "PRINTF_COND_"}> : () -> ()
  "emit.fragment"() <{sym_name = "PRINTF_COND_FRAGMENT"}> ({
    "sv.verbatim"() <{format_string = "\0A// Users can define 'PRINTF_COND' to add an extra gate to prints.", symbols = []}> : () -> ()
    "sv.ifdef"() <{cond = #sv<macro.ident @PRINTF_COND_>}> ({
    ^bb0:
    }, {
      "sv.ifdef"() <{cond = #sv<macro.ident @PRINTF_COND>}> ({
        "sv.macro.def"() <{format_string = "(`PRINTF_COND)", macroName = @PRINTF_COND_, symbols = []}> : () -> ()
      }, {
        "sv.macro.def"() <{format_string = "1", macroName = @PRINTF_COND_, symbols = []}> : () -> ()
      }) : () -> ()
    }) : () -> ()
  }) : () -> ()
  "hw.module"() <{comment = "VCS coverage exclude_file", module_type = !hw.modty<input port_0 : i64, input port_1 : i1, input port_2 : i1>, parameters = [], per_port_attrs = [], result_locs = [], sym_name = "SignedPrintETC_assert"}> ({
  ^bb0(%arg3: i64, %arg4: i1, %arg5: i1):
    %3 = "hw.constant"() <{value = -2147483646 : i32}> : () -> i32
    "sv.ifdef"() <{cond = #sv<macro.ident @SYNTHESIS>}> ({
    ^bb0:
    }, {
      "sv.always"(%arg5) <{events = [0 : i32]}> ({
        "sv.if"(%arg4) ({
          "sv.fwrite"(%3, %arg3) <{format_string = "%0d"}> : (i32, i64) -> ()
        }, {
        }) : (i1) -> ()
      }) : (i1) -> ()
    }) : () -> ()
    "hw.output"() : () -> ()
  }) {emit.fragments = [@PRINTF_COND_FRAGMENT], sym_visibility = "private"} : () -> ()
  "hw.module"() <{comment = "", module_type = !hw.modty<input clk : !seq.clock, input test : i64>, parameters = [], per_port_attrs = [], result_locs = [], sym_name = "SignedPrintETC"}> ({
  ^bb0(%arg1: !seq.clock, %arg2: i64):
    %0 = "seq.from_clock"(%arg1) : (!seq.clock) -> i1
    "sv.ifdef"() <{cond = #sv<macro.ident @SYNTHESIS>}> ({
    ^bb0:
    }, {
      "sv.always"(%0) <{events = [0 : i32]}> ({
        %1 = "sv.macro.ref.expr"() <{macroName = @PRINTF_COND_}> : () -> i1
        "sv.if"(%1) ({
          %2 = "sv.system"(%arg2) <{fnName = "signed"}> : (i64) -> i64
        }, {
        }) : (i1) -> ()
      }) : (i1) -> ()
    }) : () -> ()
    "hw.instance"(%2, %1, %0) <{argNames = ["port_0", "port_1", "port_2"], doNotPrint, inner_sym = #hw<innerSym@__ETC_SignedPrintETC_assert>, instanceName = "SignedPrintETC_assert", moduleName = @SignedPrintETC_assert, parameters = [], resultNames = []}> : (i64, i1, i1) -> ()
    "hw.output"() : () -> ()
  }) {emit.fragments = [@PRINTF_COND_FRAGMENT]} : () -> ()
  "om.class"() <{fieldNames = [], fieldTypes = {}, formalParamNames = ["basepath"], sym_name = "SignedPrintETC_Class"}> ({
  ^bb0(%arg0: !om.basepath):
    "om.class.fields"() <{field_locs = []}> : () -> ()
  }) : () -> ()
  "sv.bind"() <{instance = #hw.innerNameRef<@SignedPrintETC::@__ETC_SignedPrintETC_assert>}> : () -> ()
}) : () -> ()
```

uenoku, 2025-07-23:
seems like sv.system is blocked (https://github.com/llvm/circt/blob/5c3780f58bc220af235c5236357f7d6b964be241/lib/Dialect/SV/Transforms/SVExtractTestCode.cpp#L631) when determining operations to clone.

dtzSiFive, 2025-07-23:
Thanks for the pointer!  Probably hacking this in as well works? Signed specifically seems certainly safe to clone, dunno about all system tasks? OTOH I think we're hoping to nuke this code "soon", and we clone procedural calls too, so maybe giving that a go is the path forward.

I'm rebuilding all the things presently else I'd try that right now.

dtzSiFive, 2025-07-23:
Yes, this issue is resolved if we stop using ETC 👍 .

dtzSiFive, 2025-07-23:
Thanks, your suggestions make sense to me!

seldridge, 2025-07-23:
Dedicated operation seems fine. This is annoying as `$signed` is pedantically a system function according to the spec. However, it doesn't mean that we need to have no built-in system functions (which is mostly where we're at now).

Specifically, I'd propose `sv.system.signed` similar to how we have `sv.system.time` and friends.

dtzSiFive, 2025-07-24:
After looking at the SV specification I think we might have a few possibly better options:

1) Spec refers to system tasks as well as system functions.  Functions return one value as computation using inputs nothing more, whereas tasks can do all kinds of things.  Having this classification reflected in the IR would help here and elsewhere, perhaps one operation for system tasks and another for kind (if we prefer it as a bit or computed value of the operation, need to make sure that bridges nicely into the generic checks for side-effects).  This could get us good behavior throughout pipeline for many system functions without much effort (?).

2) For the specific situation here, I noticed SV actually has cast operators that we don't model and appear to be preferred over the Verilog system function.  They look to be defined to be equivalent in functionality.  Leveraging this we could add a cast operation (or at least signed/unsigned cast operations given how our types currently work? Not 100% on the specific IR design here, thoughts/suggestions welcome or on a PR if we go this route).  So instead of `$signed(...)` we'd emit `signed'(...)`.
If nothing else signed/unsigned cast operations might "feel" better (and avoid having multiple ways to express same thing, as with a generic system operation + various specific operations for specific system tasks/functions? I haven't looked around at what of these we use/have operations for...).

Of course we could decide it's nice to do both (1) and (2) as part of general improvements...

Any thoughts/preferences here?

It's not great but also did the simple thing and checked for signed/unsigned specifically here to workaround this issue in case that's urgent as a temporary stopgap.  See #8775.

dtzSiFive, 2025-07-28:
Looks like functions can have side-effects, I misread.  `$urandom` for example 👍 .

Furthermore we've already gone with having redundant representations for system tasks/functions (`sv.system "time" ...`; `sv.system.time`).  Assuming latter is at least a refinement of the former, that's not ideal but not dangerous beyond possibly leading to less-than-ideal output (and, FWIW, verification).

Maybe I'd feel better if we canonicalized to the latter? 👍 .

Additionally, it may be the case that `signed'` may cause compatibility concerns -- this is not based on anything beyond anything might do so 😉 .

As an aside, can we really say `$sampled` is Pure? 🤔 

Sounds like going with `sv.system.signed` as Pure is the short-term answer here.