llvm/circt #8832: [LLHD] Dominance issue with local signal

fabianschuiki, 2025-08-08:
The following input produces a dominance issue in Mem2Reg:
```mlir
// circt-opt --llhd-mem2reg
hw.module @Foo() {
  %0 = llhd.constant_time <0ns, 0d, 1e>
  %c0_i8 = hw.constant 0 : i8
  %c0_i4 = hw.constant 0 : i4
  %c0_i32 = hw.constant 0 : i32
  %1 = hw.bitcast %c0_i8 : (i8) -> !hw.array<2xi4>
  %a = llhd.sig %1 : !hw.array<2xi4>
  %2 = llhd.prb %a : !hw.inout<array<2xi4>>
  llhd.process {
    cf.br ^bb2(%c0_i32 : i32)
  ^bb1:  // pred: ^bb8
    cf.br ^bb2(%c0_i32 : i32)
  ^bb2(%3: i32):  // 3 preds: ^bb0, ^bb1, ^bb7
    %false = hw.constant false
    cf.cond_br %false, ^bb3, ^bb8
  ^bb3:  // pred: ^bb2
    %c0_i2 = hw.constant 0 : i2
    %out = llhd.sig %c0_i4 : i4
    cf.br ^bb4(%c0_i32 : i32)
  ^bb4(%4: i32):  // 2 preds: ^bb3, ^bb5
    cf.cond_br %false, ^bb5, ^bb6
  ^bb5:  // pred: ^bb4
    %5 = llhd.sig.extract %out from %c0_i2 : (!hw.inout<i4>) -> !hw.inout<i1>
    llhd.drv %5, %false after %0 : !hw.inout<i1>
    cf.br ^bb4(%c0_i32 : i32)
  ^bb6:  // pred: ^bb4
    cf.br ^bb7(%c0_i4 : i4)
  ^bb7(%6: i4):  // pred: ^bb6
    cf.br ^bb2(%c0_i32 : i32)
  ^bb8:  // pred: ^bb2
    llhd.wait (%c0_i4, %2 : i4, !hw.array<2xi4>), ^bb1
  }
  hw.output
}
```
Error:
```
broken-dom-reduced-5.mlir:19:14: error: operand #0 does not dominate this use
      %out = llhd.sig %c0_i4 : i4
             ^
broken-dom-reduced-5.mlir:19:14: note: see current operation: %6 = "llhd.prb"(%10) : (!hw.inout<i4>) -> i4
broken-dom-reduced-5.mlir:19:14: note: operand defined here (op in the same region)
```
@mvpant You have been fixing a similar issue recently. Do you know what might be wrong here?

mvpant, 2025-08-19:
@fabianschuiki Looks like you fixed this in #8859.

mvpant, 2025-08-22:
> I'm wondering whether that actually fixed it, or whether it just hides the issue because the local variable gets deleted in most situations. What do you think? 🤔 

I removed the `removeUnusedLocalSignals` call to verify whether issue reappears when signal op remains in place. Everything seems fine, lattice looks ok, no more probes sinking into the entry block.
https://github.com/llvm/circt/blob/472531b63ec62573faa9314daf02bdd6295da36e/lib/Dialect/LLHD/Transforms/Mem2Reg.cpp#L1086-L1091

> For example, what if we call a function with a reference to a local variable somewhere in there, like `func.call @use_ref_i4(%out) : (!hw.inout<i4>) -> ()`. That would block signal removal, and might uncover the dominance issue.

Unfortunately can't do this, we are limited by:
https://github.com/llvm/circt/blob/472531b63ec62573faa9314daf02bdd6295da36e/lib/Dialect/LLHD/Transforms/Mem2Reg.cpp#L809

There is a testcase that used to segfault in pre #8693 version, but now works too.
```
hw.module @Foo(in %clock: i1) {
  %0 = llhd.constant_time <0ns, 0d, 1e>
  %c0_i8 = hw.constant 0 : i8
  %c0_i4 = hw.constant 0 : i4
  %c1_i4 = hw.constant 1 : i4
  %c2_i4 = hw.constant 2 : i4
  %c0_i32 = hw.constant 0 : i32
  %1 = hw.bitcast %c0_i8 : (i8) -> !hw.array<2xi4>
  %a = llhd.sig %1 : !hw.array<2xi4>
  %2 = llhd.prb %a : !hw.inout<array<2xi4>>
  llhd.process {
    cf.br ^bb2(%c0_i4 : i4)
  ^bb1:
    cf.br ^bb2(%c0_i4 : i4)
  ^bb2(%3: i4):
    %104 = comb.extract %3 from 0 : (i4) -> i1
    %105 = comb.and %clock, %104 : i1
    %false = hw.constant false
    %true = hw.constant true
    cf.cond_br %true, ^bb3(%105 : i1), ^bb8
  ^bb3(%9: i1):
    %110 = comb.replicate %9 : (i1) -> (i4)
    %out = llhd.sig %c0_i4 : i4
    cf.br ^bb4(%c0_i32 : i32)
  ^bb4(%4: i32):
    cf.cond_br %false, ^bb5, ^bb6
  ^bb5:
    %100 = llhd.prb %out : !hw.inout<i4>
    %c0_i2 = hw.constant 0 : i2
    %5 = llhd.sig.extract %out from %c0_i2 : (!hw.inout<i4>) -> !hw.inout<i1>
    llhd.drv %5, %false after %0 : !hw.inout<i1>
    %101 = comb.icmp eq %100, %c0_i4 : i4
    %102 = comb.and %101, %clock : i1
    cf.cond_br %102, ^bb6, ^bb7
  ^bb6:
    llhd.drv %out, %c0_i4 after %0 : !hw.inout<i4>
    cf.br ^bb7
  ^bb7:
    %107 = llhd.prb %out : !hw.inout<i4>
    llhd.drv %out, %c1_i4 after %0 : !hw.inout<i4>
    cf.br ^bb1
  ^bb8:
    llhd.wait (%c0_i4, %2 : i4, !hw.array<2xi4>), ^bb1
  }
  hw.output
}
```

Overall, i think the issue is fixed. Any test i create is limited to a CFG where the local signal dominates its users (otherwise we'd get a parse error). The mentioned code removes needed defs from the IN set of signal`s basic block, which then propagates to all predecessors. I am curious to see a failling example 🤔

fabianschuiki, 2025-08-22:
Really cool, thanks for double-checking! 🥳 In that case I'll mark this as fixed by #8859.

closed 2025-08-22 by fabianschuiki