llvm/circt #8833: [Seq] Register symbol with special characters does not round-trip

fabianschuiki, 2025-08-08:
It looks like a `sym @...` on a `seq.compreg` (and likely any other op that can have an inner symbol) gets printed without `"..."` even if it contains special characters:
```
// foo.mlir
hw.module @Foo(in %a: i42, in %clk: !seq.clock) {
  %0 = seq.compreg sym @"foo/bar" %a, %clk : i42
  hw.output
}

// after circt-opt foo.mlir
hw.module @Foo(in %a : i42, in %clk : !seq.clock) {
  %0 = seq.compreg sym @foo/bar %a, %clk : i42
  hw.output
}

// after circt-opt foo.mlir | circt-opt
<stdin>:3:30: error: expected SSA operand
    %0 = seq.compreg sym @foo/bar %a, %clk : i42
                             ^
```

thomasnormal, 2026-03-03:
Reproduced on current  and fixed in my fork at commit .

Root cause:
-  and  emitted raw symbol text, so names requiring quotes (e.g. ) printed without proper symbol escaping.

Fix:
- Switched both printers in  to use .

Regression test:
- 
- Verifies  survives .

Validation:
- -- Testing: 3 tests, 1 workers --
Testing:  0.. 10.. 20.. 30.. 40.. 50.. 60.. 70.. 80.. 90..

Testing Time: 0.05s

Total Discovered Tests: 3
  Passed: 3 (100.00%) (3/3 pass).

thomasnormal, 2026-03-03:
Update: landed on `thomasnormal/circt` `main` in commit `36ae5d5e8c`.

What landed:
- `lib/Dialect/HW/HWAttributes.cpp`
  - `InnerSymAttr` / `InnerSymPropertiesAttr` printing now preserves parser-safe symbol formatting for names requiring quoting
- `test/Dialect/Seq/inner-sym-round-trip.mlir`
  - regression for `seq.compreg sym @"foo/bar"` round-trip under `--verify-roundtrip`

Validation:
- `build_test/bin/llvm-lit -sv -j1 test/Dialect/Seq/inner-sym-round-trip.mlir test/Dialect/Seq/round-trip.mlir test/Dialect/HW/basic.mlir`
- result: `3 passed`

seldridge, 2026-03-06:
@thomasnormal, @thomasahle: Due to current CIRCT project policy, we are unable to accept patches from the `thomasnormal/circt` fork.

For others: please do not upstream any patches from the `thomasnormal/circt` fork. Additionally, recognize that any comments made by the @thomasnormal account appear to be from one or more AI agents.

For more information, please see this Discourse post: https://discourse.llvm.org/t/circt-project-policy-on-contributions-from-thomasnormal-circt-fork/90075