llvm/circt #8845: [circt-verilog] `circt-verilog` produces non `comb`/`seq` dialects including `cf` and `llhd`

AtticusKuhn, 2025-08-12:
I am trying to convert this file found in OpenTitan from SystemVerilog to MLIR:
https://github.com/lowRISC/opentitan/blob/d8b5efd1427152b8387d6e03d9db413167e58475/hw/top_englishbreakfast/ip_autogen/pwrmgr/rtl/pwrmgr_fsm.sv

When I run `circt-verilog`, it produces unexpected dialects like `cf` and `llhd`. I don't expect these dialects to be in the output.

Example:
```bash
ak2518@autobot:/local/scratch/ak2518/opentitan$ /local/scratch/ak2518/circt/build/bin/circt-verilog /local/scratch/ak2518/opentitan/hw/top_englishbreakfast/ip_autogen/pwrmgr/rtl/pwrmgr_fsm.sv -yhw/ip/prim_generic/rtl -yhw/ip/prim/rtl -yhw/top_englishbreakfast/ip_autogen/pwrmgr/rtl -yhw/ip/lc_ctrl/rtl -yhw/ip/rom_ctrl/rtl -UINC_ASSERT -Ihw/ip/prim/rtl -DYOSYS -DSYNTHESIS  | grep "llhd\."
    %92:19 = llhd.combinational -> i1, i1, i1, i1, i1, i1, i1, i1, i1, i12, i1, i1, i1, i1, i2, i2, i2, i1, i4 {
      llhd.yield %178, %179, %180, %181, %182, %183, %184, %185, %186, %198, %187, %188, %189, %190, %191, %192, %193, %194, %195 : i1, i1, i1, i1, i1, i1, i1, i1, i1, i12, i1, i1, i1, i1, i2, i2, i2, i1, i4
```

```bash
ak2518@autobot:/local/scratch/ak2518/opentitan$ /local/scratch/ak2518/circt/build/bin/circt-verilog /local/scratch/ak2518/opentitan/hw/top_englishbreakfast/ip_autogen/pwrmgr/rtl/pwrmgr_fsm.sv -yhw/ip/prim_generic/rtl -yhw/ip/prim/rtl -yhw/top_englishbreakfast/ip_autogen/pwrmgr/rtl -yhw/ip/lc_ctrl/rtl -yhw/ip/rom_ctrl/rtl -UINC_ASSERT -Ihw/ip/prim/rtl -DYOSYS -DSYNTHESIS  | grep "cf\."
      cf.cond_br %95, ^bb1(%req_pwrup_i, %reset_ongoing_q, %reset_cause_q, %u_fetch_en.lc_en_o, %c-1330_i12, %u_fetch_en.lc_en_o : i1, i1, i2, i4, i12, i4), ^bb2
      cf.cond_br %102, ^bb32(%false, %false, %false, %false, %false, %false, %false, %false, %false, %ack_pwrup_q, %req_pwrdn_q, %reset_ongoing_q, %ip_clk_en_q, %rst_lc_req_q, %rst_sys_req_q, %98, %low_power_q, %99, %100 : i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i2, i2, i2, i1, i4, i12), ^bb32(%false, %false, %false, %false, %false, %false, %false, %false, %false, %ack_pwrup_q, %req_pwrdn_q, %reset_ongoing_q, %ip_clk_en_q, %rst_lc_req_q, %rst_sys_req_q, %reset_cause_q, %low_power_q, %101, %u_state_regs.state_o : i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i2, i2, i2, i1, i4, i12)
      cf.cond_br %103, ^bb3(%44, %false, %false, %req_pwrdn_q, %true, %rst_lc_req_q, %rst_sys_req_q, %reset_cause_q, %c-1600_i12, %false, %req_pwrdn_q : i1, i1, i1, i1, i1, i2, i2, i2, i12, i1, i1), ^bb4
      cf.cond_br %104, ^bb32(%105, %106, %false, %false, %false, %false, %false, %false, %false, %ack_pwrup_q, %107, %reset_ongoing_q, %108, %109, %110, %111, %low_power_q, %u_fetch_en.lc_en_o, %112 : i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i2, i2, i2, i1, i4, i12), ^bb32(%105, %113, %false, %false, %false, %false, %false, %false, %false, %ack_pwrup_q, %114, %reset_ongoing_q, %108, %109, %110, %111, %low_power_q, %u_fetch_en.lc_en_o, %u_state_regs.state_o : i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i2, i2, i2, i1, i4, i12)
      cf.cond_br %115, ^bb5, ^bb6
      cf.cond_br %116, ^bb32(%false, %false, %false, %false, %false, %false, %false, %false, %false, %ack_pwrup_q, %req_pwrdn_q, %reset_ongoing_q, %ip_clk_en_q, %c0_i2, %c0_i2, %reset_cause_q, %low_power_q, %u_fetch_en.lc_en_o, %c-94_i12
: i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i2, i2, i2, i1, i4, i12), ^bb32(%false, %false, %false, %false, %false, %false, %false, %false, %false, %ack_pwrup_q, %req_pwrdn_q, %reset_ongoing_q, %ip_clk_en_q, %c0_i2, %c0_i2, %reset_cause_q, %low_power_q, %u_fetch_en.lc_en_o, %u_state_regs.state_o : i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i2, i2, i2, i1, i4, i12)
      cf.cond_br %117, ^bb3(%otp_done_i, %true, %false, %req_pwrdn_q, %ip_clk_en_q, %rst_lc_req_q, %rst_sys_req_q, %reset_cause_q, %c-1453_i12, %false, %req_pwrdn_q : i1, i1, i1, i1, i1, i2, i2, i2, i12, i1, i1), ^bb7
      cf.cond_br %118, ^bb3(%u_sync_lc_done.q_o, %false, %true, %req_pwrdn_q, %ip_clk_en_q, %rst_lc_req_q, %rst_sys_req_q, %reset_cause_q, %c168_i12, %true, %req_pwrdn_q : i1, i1, i1, i1, i1, i2, i2, i2, i12, i1, i1), ^bb8
      cf.cond_br %119, ^bb9, ^bb11
      cf.cond_br %122, ^bb10, ^bb32(%false, %false, %false, %false, %false, %false, %false, %false, %false, %120, %req_pwrdn_q, %reset_ongoing_q, %ip_clk_en_q, %rst_lc_req_q, %rst_sys_req_q, %reset_cause_q, %low_power_q, %u_fetch_en.lc_en_o, %u_state_regs.state_o : i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i2, i2, i2, i1, i4, i12)
      cf.br ^bb32(%false, %false, %124, %false, %false, %false, %true, %false, %false, %false, %req_pwrdn_q, %false, %ip_clk_en_q, %rst_lc_req_q, %rst_sys_req_q, %reset_cause_q, %low_power_q, %u_fetch_en.lc_en_o, %c-966_i12 : i1, i1, i1,
i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i2, i2, i2, i1, i4, i12)
      cf.cond_br %125, ^bb12, ^bb13
      cf.br ^bb32(%false, %false, %false, %false, %false, %false, %false, %126, %false, %ack_pwrup_q, %req_pwrdn_q, %reset_ongoing_q, %ip_clk_en_q, %rst_lc_req_q, %rst_sys_req_q, %reset_cause_q, %low_power_q, %u_fetch_en.lc_en_o, %c1523_i12 : i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i2, i2, i2, i1, i4, i12)
      cf.cond_br %127, ^bb14(%82, %c0_i2, %false, %c1028_i12 : i4, i2, i1, i12), ^bb15
      cf.cond_br %132, ^bb32(%false, %false, %false, %false, %false, %false, %false, %false, %false, %ack_pwrup_q, %req_pwrdn_q, %reset_ongoing_q, %ip_clk_en_q, %rst_lc_req_q, %rst_sys_req_q, %129, %130, %u_fetch_en.lc_en_o, %131 : i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i2, i2, i2, i1, i4, i12), ^bb32(%false, %false, %false, %false, %false, %false, %false, %false, %false, %ack_pwrup_q, %req_pwrdn_q, %reset_ongoing_q, %ip_clk_en_q, %rst_lc_req_q, %rst_sys_req_q, %129, %130, %u_fetch_en.lc_en_o, %u_state_regs.state_o : i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i2, i2, i2, i1, i4, i12)
      cf.cond_br %133, ^bb14(%91, %reset_cause_q, %low_power_q, %c868_i12 : i4, i2, i1, i12), ^bb16
      cf.cond_br %134, ^bb1(%19, %low_power_entry_i, %c-1_i2, %c-6_i4, %c917_i12, %c5_i4 : i1, i1, i2, i4, i12, i4), ^bb17
      cf.cond_br %135, ^bb18, ^bb19
      cf.br ^bb32(%false, %false, %false, %false, %false, %false, %false, %false, %false, %ack_pwrup_q, %req_pwrdn_q, %reset_ongoing_q, %false, %rst_lc_req_q, %rst_sys_req_q, %reset_cause_q, %139, %u_fetch_en.lc_en_o, %138 : i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i2, i2, i2, i1, i4, i12)
      cf.cond_br %140, ^bb20, ^bb21
      cf.cond_br %141, ^bb32(%false, %false, %true, %true, %false, %true, %false, %false, %false, %ack_pwrup_q, %req_pwrdn_q, %reset_ongoing_q, %true, %rst_lc_req_q, %rst_sys_req_q, %reset_cause_q, %low_power_q, %u_fetch_en.lc_en_o, %c1523_i12 : i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i2, i2, i2, i1, i4, i12), ^bb32(%false, %false, %false, %false, %false, %true, %false, %false, %false, %ack_pwrup_q, %req_pwrdn_q, %reset_ongoing_q, %ip_clk_en_q, %rst_lc_req_q,
%rst_sys_req_q, %reset_cause_q, %low_power_q, %u_fetch_en.lc_en_o, %c-1671_i12 : i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i2, i2, i2, i1, i4, i12)
      cf.cond_br %142, ^bb22, ^bb23
      cf.cond_br %143, ^bb32(%false, %false, %false, %false, %false, %false, %false, %false, %false, %ack_pwrup_q, %req_pwrdn_q, %reset_ongoing_q, %ip_clk_en_q, %rst_lc_req_q, %rst_sys_req_q, %reset_cause_q, %low_power_q, %u_fetch_en.lc_en_o, %c1423_i12 : i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i2, i2, i2, i1, i4, i12), ^bb32(%false, %false, %true, %false, %true, %false, %false, %false, %false, %ack_pwrup_q, %req_pwrdn_q, %reset_ongoing_q, %true, %rst_lc_req_q, %rst_sys_req_q, %reset_cause_q, %low_power_q, %u_fetch_en.lc_en_o, %c1523_i12 : i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i2, i2, i2, i1, i4, i12)
      cf.cond_br %144, ^bb24(%c1_i32, %rst_lc_req_q, %rst_sys_req_q : i32, i2, i2), ^bb26
      cf.cond_br %148, ^bb25, ^bb3(%30, %false, %false, %req_pwrdn_q, %ip_clk_en_q, %146, %147, %c1_i2, %c-375_i12, %false, %req_pwrdn_q : i1, i1, i1, i1, i1, i2, i2, i2, i12, i1, i1)
      cf.br ^bb24(%163, %160, %162 : i32, i2, i2)
      cf.cond_br %164, ^bb3(%ack_pwrdn_i, %false, %false, %false, %ip_clk_en_q, %rst_lc_req_q, %rst_sys_req_q, %reset_cause_q, %c55_i12, %false, %true : i1, i1, i1, i1, i1, i2, i2, i2, i12, i1, i1), ^bb27
      cf.cond_br %165, ^bb32(%false, %false, %false, %false, %false, %true, %false, %false, %false, %ack_pwrup_q, %req_pwrdn_q, %true, %ip_clk_en_q, %rst_lc_req_q, %rst_sys_req_q, %reset_cause_q, %low_power_q, %u_fetch_en. [... cut ...]  %low_power_q, %u_fetch_en.lc_en_o, %u_state_regs.state_o : i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i2, i2, i2, i1, i4, i12)
```

I do not expect this behavior. I expect `circt-verilog` to produce code in the `seq`/`comb` dialects.

Version:
```bash
ak2518@autobot:/local/scratch/ak2518/opentitan$ /local/scratch/ak2518/circt/build/bin/circt-verilog --version
LLVM (http://llvm.org/):
  LLVM version 22.0.0git
  DEBUG build with assertions.
CIRCT 55f12edd1
slang version 9.0.0+54385bb
```

fabianschuiki, 2025-08-28:
Fantastic! Thanks for reducing this down so thoroughly, @AtticusKuhn.

seldridge, 2026-03-06:
@thomasnormal, @thomasahle: Due to current CIRCT project policy, we are unable to accept patches from the `thomasnormal/circt` fork.

For others: please do not upstream any patches from the `thomasnormal/circt` fork. Additionally, recognize that any comments made by the @thomasnormal account appear to be from one or more AI agents.

For more information, please see this Discourse post: https://discourse.llvm.org/t/circt-project-policy-on-contributions-from-thomasnormal-circt-fork/90075