llvm/circt #8925: [Comb] Fold comb.truth_table when truth table depends on a single operand

uenoku, 2025-09-05:
```
  hw.module @Foo(in %c : i3, out out : i1) {
    %0 = comb.extract %c from 2 : (i3) -> i1
    %1 = comb.extract %c from 1 : (i3) -> i1
    %2 = comb.extract %c from 0 : (i3) -> i1
    %3 = comb.truth_table %2, %0, %1 -> [false, false, false, false, true, true, true, true]
    hw.output %3 : i1
  }
```
`%3` should be canonicalized to `%2`. So we should be able to canonicalize the op if the truth table depends on a single variable:
```
00001111
00110011
01010101
```

kart-u, 2025-10-05:
hello @uenoku 
can this be assigned to me?, I would love to contribute!

uenoku, 2025-10-05:
Please do!

kart-u, 2025-10-05:
Thank you for assigning this to me,
I have started working on this had some doubt on my understanding
here you showed that there is no dependence on %0 and %1 thus for %3 we should decide from %2 only
(means there are unused operands and we can optimize)

so my approach is I will change CombFolds.cpp and write a function to remove unused operands 
am I in right direction??

kart-u, 2025-10-08:
I have written change in code but due build problems (my laptop is low-end thus building taking too much time or in some cases failing in between) I am unable to test those changes , and finding hard to work in this discontinues manner , you can assign this issue  to someone else, Sorry for inconvenience

closed 2025-11-27 by uenoku