llvm/circt #8973: [MooreToCore] Lowering to math.ipow?

cowardsa, 2025-09-16:
Is there a strong reason to go from PowUOp to math.ipow as opposed to lowering to a sequence of multiplications?

Is there any existing lowering from math.ipow to the core dialects? 

https://github.com/llvm/circt/blob/83effb21cd6b1cebc060b4f57180f16683309cd9/lib/Conversion/MooreToCore/MooreToCore.cpp#L1376

`
struct PowUOpConversion : public OpConversionPattern<PowUOp> {
  using OpConversionPattern::OpConversionPattern;

  LogicalResult
  matchAndRewrite(PowUOp op, OpAdaptor adaptor,
                  ConversionPatternRewriter &rewriter) const override {
    Type resultType = typeConverter->convertType(op.getResult().getType());

    Location loc = op->getLoc();

    Value zeroVal = hw::ConstantOp::create(rewriter, loc, APInt(1, 0));
    // zero extend both LHS & RHS to ensure the unsigned integers are
    // interpreted correctly when calculating power
    auto lhs = comb::ConcatOp::create(rewriter, loc, zeroVal, adaptor.getLhs());
    auto rhs = comb::ConcatOp::create(rewriter, loc, zeroVal, adaptor.getRhs());

    // lower the exponentiation via MLIR's math dialect
    auto pow = mlir::math::IPowIOp::create(rewriter, loc, lhs, rhs);

    rewriter.replaceOpWithNewOp<comb::ExtractOp>(op, resultType, pow, 0);
    return success();
  }
};
`

cowardsa, 2025-09-17:
Agree with Fabian that it's very division like - but hadn't thought about simulation. Perhaps we need both - constant powers lowered to multiplication and non-constants lowered to comb. 

How is division by a power of two handled? (feels quite similar) Is it a canonicalization pattern?

fabianschuiki, 2025-09-17:
I don't see any canonicalizations in `CombFolds.cpp` for `comb.divu` or `comb.divs`. We only have folders for trivial division by one. But we also don't seem to handle divisions by constants, and divisions by powers-of-two for that matter. Might make sense to turn those into shifts.

uenoku, 2025-09-19:
FYI regarding CombToSynth an unsigned division by a power of two is lowered into a shift, otherwise it's lowered to a truth-table if unknown input bits are less than a constant specified by a pass option.