llvm/circt #9013: [circt-opt] Segmentation fault during XOR op building

likeamahoney, 2025-09-24:
Hi, all! Let's consider a such `Verilog` example:

```verilog
module top (input clk, input rstn);
    reg r1;
    always @(posedge clk or negedge rstn) begin
        r1 = 0;
    end
endmodule
```

During `llhd-desequentialize` pass `circt-opt` tool segfaults using this pipeline:

```
circt-verilog test.sv | circt-opt --llhd-desequentialize
```

With such stacktrace:

```
```

fzi-hielscher, 2025-09-26:
Hi @likeamahoney,

I hope @fabianschuiki or @maerhart will correct me if I'm wrong. As far as I know `--llhd-desequentialize` has been deprecated and replaced by `--lhd-deseq` which appears to be handling your example just fine. Is there a reason you are specifically invoking `--llhd-desequentialize`?

likeamahoney, 2025-09-26:
> Hi [@likeamahoney](https://github.com/likeamahoney),
> 
> I hope [@fabianschuiki](https://github.com/fabianschuiki) or [@maerhart](https://github.com/maerhart) will correct me if I'm wrong. As far as I know `--llhd-desequentialize` has been deprecated and replaced by `--llhd-deseq` which appears to be handling your example just fine. Is there a reason you are specifically invoking `--llhd-desequentialize`?

Hi, I have't got any specific reason for running this pass instead of `--llhd-deseq`. I just tried some combinations of pass to lower my example to `arc`  and `llvm` and during this have found it.

So If it is deprecated I can close this issue.

fabianschuiki, 2025-09-26:
Nice find @likeamahoney! 😃 The fact that `circt-verilog` spits out an `llhd.process` means that the `llhd-deseq` pass ran as part of the circt-verilog pipeline didn't properly recognize this process as a register. That is definitely a bug. @fzi-hielscher is also right, the `llhd-desequentialize` pass is deprecated. We should just delete that at this point, since the pipeline has been running pretty well for a few months now with only the updated passes.

I think the thing that trips up the Deseq pass in your example is the fact that its using a blocking assignment `r1 = 0` instead of a non-blocking assignment `r1 <= 0` in the `always` process. Does this come from some real-world Verilog source code? The issue with a non-blocking assignment is that it would happen immediately when the `always` process is triggered by a clock edge, instead of the usual propagation delay of 1 delta time step that tries to emulate register propagation times in hardware. If you had multiple registers with `=` assignments that use each other's values, this would create a race condition where some registers would already see the updated value of other registers. The delta time step is necessary to give all `always` processes in the design a chance to trigger and read the old values of their input registers, and then have all register changes apply 1 delta time step worth of propagation delay later.

likeamahoney, 2025-09-27:
> Nice find [@likeamahoney](https://github.com/likeamahoney)! 😃 The fact that `circt-verilog` spits out an `llhd.process` means that the `llhd-deseq` pass ran as part of the circt-verilog pipeline [...]

Thank you for the answer!

I got this example from one of my previous issue -.

Yes blocking assignments it is not the best way to design sequential logic but - but such kind of example are presence in real-world open and proprietary design. For example famous `picorv32` which was synthesed many times - https://github.com/YosysHQ/picorv32/blob/87c89acc18994c8cf9a2311e871818e87d304568/picorv32.v#L1406.

fabianschuiki, 2025-09-29:
Nice, thanks! I'm wondering how we should deal with such inputs which are subtly wrong. There are a few other common Verilog design issues that `circt-verilog` already works around:

- treat `always @*` as `always_comb`, because people often don't understand that with `always @*` the process is not executed on the initial values of the input signals
- assume that a `case ... endcase` statement which covers all two-valued possibilities is meant to be covering all four-valued possibilities (this one is wild and wrong, but people rely on it in the real world)

We could think about treating `=` blocking assignments in an `always` process with edge sensitivity, e.g. `posedge clk`, as a non-blocking assignment for the sake of register detection. We should probably make that a lowering option, or something we detect and fixup as a pass at the Moore dialect level. The user should then be able to disable this fixup if they want accurate Verilog compliance. WDYT?

likeamahoney, 2025-10-03:
> RHS of the non-blocking assignment is evaluated when the assignment is encountered

yes, sorry, I forgot this fact from LRM

>  But as you've pointed out, some Verilog code in the wild uses exactly this and expects a register to result from this.

I try to synthesis this model example on `yosys` to check what would be produced after optimizations and logical synthesis:

```verilog
module m(input clk, input c, output reg a, output reg b);

always @(posedge clk) begin
 a = c;
 b <= a;
 a = a + 1;
end

endmodule
```

below result of synthesis is presented:

```
module \m
  attribute \force_downto 1
  attribute \src "1.v:6.6-6.11|/usr/bin/../share/yosys/techmap.v:270.23-270.24"
  wire $auto$alumacc.cc:485:replace_alu$5.X
  attribute \src "1.v:1.41-1.42"
  wire output 3 \a
  attribute \src "1.v:1.55-1.56"
  wire output 4 \b
  attribute \src "1.v:1.27-1.28"
  wire input 2 \c
  attribute \src "1.v:1.16-1.19"
  wire input 1 \clk
  cell $_NOT_ $abc$105$auto$blifparse.cc:386:parse_blif$106
    connect \A \c
    connect \Y $auto$alumacc.cc:485:replace_alu$5.X
  end
  attribute \src "1.v:3.1-7.4"
  cell $_DFF_P_ $auto$ff.cc:266:slice$90
    connect \C \clk
    connect \D $auto$alumacc.cc:485:replace_alu$5.X
    connect \Q \a
  end
  attribute \src "1.v:3.1-7.4"
  cell $_DFF_P_ $auto$ff.cc:266:slice$91
    connect \C \clk
    connect \D \c
    connect \Q \b
  end
end
```

and surely here we have a 2 ffs (2 clocked 1'bit registers using other words)

> Maybe we could just make the Deseq pass also lower blocking assignments like this to registers if an option is set?

so thats why I suppose it make sense

fabianschuiki, 2025-10-03:
I just tried to run that example module `m` of yours through `circt-verilog`, and it produces the following:
```mlir
hw.module @m(in %clk : i1, in %c : i1, out a : i1, out b : i1) {
  %true = hw.constant true
  %0 = comb.add %c, %true : i1
  %1 = seq.to_clock %clk
  %a = seq.firreg %0 clock %1 : i1
  %b = seq.firreg %c clock %1 : i1
  hw.output %a, %b : i1, i1
}
```
That looks pretty much exactly like what Yosys produces. What do you think?

likeamahoney, 2025-10-04:
> I just tried to run that example module `m` of yours through `circt-verilog`, and it produces the following:
> 
> hw.module @m(in %clk : i1, in %c : i1, out a : i1, out b : i1) {
>   %true = hw.constant true
>   %0 = comb.add %c, %true : i1
>   %1 = seq.to_clock %clk
>   %a = seq.firreg %0 clock %1 : i1
>   %b = seq.firreg %c clock %1 : i1
>   hw.output %a, %b : i1, i1
> }
> That looks pretty much exactly like what Yosys produces. What do you think?

Yes, It looks very similar with regs on `a` and `b` and comb logic on `c` . `b` updated directly with `c` value on the start of clock cycle.

thomasnormal, 2026-03-03:
Rechecked this on current downstream workspace.\n\nThe deprecated pass from the report is no longer available:\n- `build_test/bin/circt-opt --llhd-desequentialize`\n- current result: `Unknown command line argument '--llhd-desequentialize'. Did you mean '--llhd-deseq'?`\n\nSo the original crash path through `--llhd-desequentialize` is not user-reachable anymore on current builds.